spla-tam/SplaTAM #1: Error creating environment

davidceka, 2023-12-05:
Hi! I'm having some problems installing the chamferdist module:
`      In file included from chamferdist/knn.cu:4:
      /home/david/miniconda3/envs/splatam/lib/python3.10/site-packages/torch/include/ATen/cuda/CUDAContext.h:6:10: fatal error: cusparse.h: No such file or directory
          6 | #include <cusparse.h>
            |          ^~~~~~~~~~~~
      compilation terminated.
      error: command '/home/david/miniconda3/envs/splatam/bin/nvcc' failed with exit code 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for chamferdist
  Running setup.py clean for chamferdist
Failed to build chamferdist
ERROR: Could not build wheels for chamferdist, which is required to install pyproject.toml-based projects
`
I've installed cuda 11.6 by `conda install -c "nvidia/label/cuda-11.6.0" cuda-toolkit` but the error keeps that way.

Any suggestions? The error appears when trying to run the demo

Nik-V9, 2023-12-05:
Hi, Thanks for trying out our code!

I've removed the dependency for chamferdist since we don't use it. Please try the latest code and let me know!

zt-yang, 2023-12-05:
I'm on Ubuntu 22.04. I didn't encounter that error with chamferdist but with diff-gaussian-rasterization-w-depth

```
$ pip install git+https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110      
Collecting git+https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110
  Cloning https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 to /tmp/pip-req-build-pj5totch
  Running command git clone --filter=blob:none --quiet https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 /tmp/pip-req-build-pj5totch
  fatal: repository 'https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110/' not found
  error: subprocess-exited-with-error
  
  × git clone --filter=blob:none --quiet https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 /tmp/pip-req-build-pj5totch did not run successfully.
  │ exit code: 128
  ╰─> See above for output.
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× git clone --filter=blob:none --quiet https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 /tmp/pip-req-build-pj5totch did not run successfully.
│ exit code: 128
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.
```

if I try without the commit number 

```
$ pip install git+https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth                                              
Collecting git+https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth
  Cloning https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth to /tmp/pip-req-build-tpbcmtjs
  Running command git clone --filter=blob:none --quiet https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth /tmp/pip-req-build-tpbcmtjs
  Resolved https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth to commit cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110
  Running command git submodule update --init --recursive -q
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: diff-gaussian-rasterization
  Building wheel for diff-gaussian-rasterization (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [53 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-310
      creating build/lib.linux-x86_64-cpython-310/diff_gaussian_rasterization
      copying diff_gaussian_rasterization/__init__.py -> build/lib.linux-x86_64-cpython-310/diff_gaussian_rasterization
      running build_ext
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-req-build-tpbcmtjs/setup.py", line 17, in <module>
          setup(
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/wheel/bdist_wheel.py", line 368, in run
          self.run_command("build")
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
[... 7 lines cut ...]
          self.distribution.run_command(command)
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/command/build_ext.py", line 88, in run
          _build_ext.run(self)
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 345, in run
          self.build_extensions()
        File "/home/yang/miniconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 434, in build_extensions
[... cut ...]
      The detected CUDA version (12.3) mismatches the version that was used to compile
      PyTorch (11.6). Please make sure to use the same CUDA versions.
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for diff-gaussian-rasterization
  Running setup.py clean for diff-gaussian-rasterization
Failed to build diff-gaussian-rasterization
ERROR: Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-based projects
```

zt-yang, 2023-12-05:
Seems like cudatoolkit 11.6 can't be installed on 22.04
[image]

davidceka, 2023-12-05:
> I'm on Ubuntu 22.04. I didn't encounter that error with chamferdist but with diff-gaussian-rasterization-w-depth
>
> ```
> $ pip install [...]

You have installed cuda 12.3, while the build needs 11.6

> Hi, Thanks for trying out our code!
> 
> I've removed the dependency for chamferdist since we don't use it. Please try the latest code and let me know!

Thanks for the prompt response! Now the code successfully works, but the app doesn't connect to the script even if in the same network. Any additional things to do?

Nik-V9, 2023-12-05:
Thanks for this discussion! We'll further update the Readme to make installation seamless. @JayKarhade 

I am going to open another issue for the connection issue:

Regarding App Connection: I've found that some networks don't allow the two devices to connect (for example, CMU's WiFi). However, using a router with CMU's internet worked. Also, I've found that it's good to clean out the app from the background when connecting it to the Python script.

I've seen unexplained connection issues with the NeRFCapture App. (https://github.com/jc211/NeRFCapture/issues/11)

We will try to release a better app and script to replace this for better support and to enable video streaming soon.

closed 2023-12-05 by Nik-V9