spla-tam/SplaTAM #5: online demo issue

H-tr, 2023-12-07:
Hi! Thanks for your impressive work.

When I'm trying to use the online demo, I got the following error:
```bash
Waiting for frames...
1/10 frames received
1/10 frames received
Traceback (most recent call last):
  File "/home/run/Data/SplaTAM/scripts/iphone_demo.py", line 561, in <module>
    dataset_capture_loop(reader, Path(config['workdir']), config['overwrite'], 
  File "/home/run/Data/SplaTAM/scripts/iphone_demo.py", line 156, in dataset_capture_loop
    images_dir.mkdir()
  File "/home/run/miniconda3/envs/splatam/lib/python3.10/pathlib.py", line 1175, in mkdir
    self._accessor.mkdir(self, mode)
FileExistsError: [Errno 17] File exists: 'experiments/iPhone_Captures/splatam_demo/rgb'
Seed set to: 0 (type: <class 'int'>)
Traceback (most recent call last):
  File "/home/run/Data/SplaTAM/viz_scripts/final_recon.py", line 297, in <module>
    visualize(scene_path, viz_cfg)
  File "/home/run/Data/SplaTAM/viz_scripts/final_recon.py", line 170, in visualize
    w2c, k = load_camera(cfg, scene_path)
  File "/home/run/Data/SplaTAM/viz_scripts/final_recon.py", line 26, in load_camera
    all_params = dict(np.load(scene_path, allow_pickle=True))
  File "/home/run/miniconda3/envs/splatam/lib/python3.10/site-packages/numpy/lib/npyio.py", line 427, in load
    fid = stack.enter_context(open(os_fspath(file), "rb"))
FileNotFoundError: [Errno 2] No such file or directory: '././experiments/iPhone_Captures/splatam_demo/SplaTAM_iPhone/params.npz'
```
If I change the `mkdir()` to `mkdir(exist_ok=True)`, the output became
```bash
Waiting for frames...
1/10 frames received
1/10 frames received
1/10 frames received
1/10 frames received
```
and keep repeating. It seems my program failed to receive continuous sequential images.

Do you have any suggestions?

Nik-V9, 2023-12-07:
Hi, Thanks for testing out the code!

The Python script is not receiving depth. If a depth image is not received, the script will skip the image.

https://github.com/spla-tam/SplaTAM/blob/b00e6eaa4fd11ea915de4749c422fce7fe10741c/scripts/iphone_demo.py#L188

Is your NeRFCapture app mentioning "depth supported" in the top right corner below connections? Please make sure to use a Pro series iPhone or a LiDAR-equipped Apple Device.

closed 2023-12-07 by H-tr

H-tr, 2023-12-07:
Thanks for your kind reply, I will close the issue.
Btw, there is a small spelling problem in README.md for downloading datasets. For example, the `download_replica.sh` should be `downloads_replica.sh` haha