spla-tam/SplaTAM #6: Is the visualization result correct?

hhcxx2006, 2023-12-07:
hi，I am very exciting to your work and want to use your code in our scene. I downloaded Replica datatset and want to test your code first. According to your illustration, I use the command "python3 viz_scripts/final_recon.py configs/replica/splatam.py". The picture below is my final visualization about room0. I find that the reconstruction result seems not correct. This is not the whole room, but only a part of room. What's wrong with this?
[image]

Nik-V9, 2023-12-07:
Hi, this seems to be a problem with the visualizer. The whole scene's camera trajectory is getting visualized. For some reason, your interactive rendering is not working.

Could you let me know if you're using the latest version of the scripts? Could you share your `params.npz` so that I could check things on my end?

hhcxx2006, 2023-12-07:
No, the `params.npz` was generated by your yesterday code. Here is the file, you can check it.
https://drive.google.com/file/d/1Py02yvSJZmlnGp3KrTkxRX3tliNtpmh7/view?usp=drive_link

Nik-V9, 2023-12-07:
Thanks for sharing the params. The visualization looks correct on my end.

[image]

This looks like an Open3D issue on your end. Are you using the correct version (0.16.0)?

Another check would be to add a print statement at this line to check if your code is entering the while loop (interactive rendering):

https://github.com/spla-tam/SplaTAM/blob/b00e6eaa4fd11ea915de4749c422fce7fe10741c/viz_scripts/final_recon.py#L243

hhcxx2006, 2023-12-07:
Hi, thank you for your help. I changed my Open3D from 0.17.0 to 0.16.0. Then it works! Thank you!
Now I want to use Intel RealSense D435i to collect my own data. I wonder what inputs should I have? I can collect rgb and depth images. Should I provide camera pose somewhere? How to use my own data?  Thank you

Nik-V9, 2023-12-08:
Closing this issue for now. If you have further queries regarding RealSense, please open another issue.

closed 2023-12-08 by Nik-V9