spla-tam/SplaTAM #8: release time for SplaTAM V2

FrontierBreaker, 2023-12-07:
Hi, thanks for your awesome work! I am wondering about the release time of SplaTAM V2 and the difference between SplaTAM V2 and SplaTAM V1. Thanks!

closed 2023-12-07 by Nik-V9

Willyzw, 2023-12-08:
Hi, thanks for sharing this amazing work. I am curious about the upcoming SplaTAM v2. Will it be an enhancement based on a new paper, or it is to represent the official implementation aligned with the performance metrics reported in the current SpanTAM paper?

Nik-V9, 2023-12-08:
Hi, Thanks for your query! The current code reflects the performance metrics reported in the paper.

SplaTAM V2 is expected to be a further refinement of the current paper and code, where we aim to increase both the speed and performance.

Nik-V9, 2024-05-01:
Hi, Thanks for the interest! It's amazing to see so much impressive follow-up work.

Given how fast follow-up work has progressed, we are holding back updates on SplaTAM for follow-up work. The new timeline should be sometime around November.