spla-tam/SplaTAM #9: Does it work on Windows?

shayse-xr, 2023-12-07:
Do you know if I can run this on Windows or do I need WSL? I tried to run the demo using Git Bash, but unfortunately it isn't working.

thx for your great work!

Nik-V9, 2023-12-07:
Hi, we haven't tested on Windows yet. 

I presume the setup would be similar to the instructions here: https://github.com/graphdeco-inria/gaussian-splatting#local-setup

Also, this video might be of help: https://youtu.be/UXtuigy_wYc?si=y87CrnBaZBPyBshL

I think the following should work if you have conda and the other CUDA 11.6 dependencies set:
```
SET DISTUTILS_USE_SDK=1 # Windows only
conda env create --file environment.yml
conda activate gaussian_splatting
```

nirmalsnair, 2023-12-09:
### Tested the iPhone online demo on Windows 10, it's working.

1. Install the conda environment as detailed [here](https://github.com/spla-tam/SplaTAM#recommended):
```
conda create -n splatam python=3.10
conda activate splatam
conda install -c "nvidia/label/cuda-11.6.0" cuda-toolkit
conda install pytorch==1.12.1 torchvision==0.13.1 torchaudio==0.12.1 cudatoolkit=11.6 -c pytorch -c conda-forge
pip install -r requirements.txt
```

2. Add `conda` to Git Bash as detailed [here](https://fmorenovr.medium.com/how-to-add-conda-to-git-bash-windows-21f5e5987f3d):
  a. Go to the folder: `Anaconda/etc/profile.d/`
  b. Run `echo ". ${PWD}/conda.sh" >> ~/.bashrc`

3. Connect your iPhone and laptop to the same network. (I used a WiFi hotspot from my laptop to connect the iPhone, since connecting both devices to a WiFi network didn't work for me.)

5. In Git Bash, activate the `splatam` (conda) environment:
```
Nirmal@LAPTOP-HEKCV3H4 MINGW64 ~
$ conda activate splatam
(splatam)
```

6. Change current directory to SplaTAM folder:
```
Nirmal@LAPTOP-HEKCV3H4 MINGW64 ~
$ cd "C:\Users\Nirmal\Documents\SplaTAM"
(splatam)
```

7. Run the iPhone online demo:
```
Nirmal@LAPTOP-HEKCV3H4 MINGW64 ~/Documents/SplaTAM (main)
$ python scripts/iphone_demo.py --config configs/iphone/online_demo.py
```

8. On the iPhone, click `Send` to capture and send a frame. (Need to capture 10 frames.)

    <details>
    <summary>Git Bash log</summary>

    ```
    Waiting for frames...
    1/10 frames received
    Tracking Result Time Step: 0: 100%|##########| 1/1 [00:00<00:00,  1.67it/s, Time-Step: 0 | Rel Pose Error: 0.0 | Pose Error: 1.490116e-08 | ATE RMSE=0.0]

    Selected Keyframes at Frame 0: [0]
    Mapping Time Step: 0: 100%|##########| 60/60 [00:02<00:00, 24.70it/s]
    Mapping Result Time Step: 0: 100%|##########| 1/1 [00:00<00:00, 46.54it/s, Time-Step: 0 | Frame 0 | PSNR: 21.96987 | Depth RMSE: 0.02397114 | L1=0.02397114]
    2/10 frames received
    Tracking Time Step: 1: 100%|##########| 60/60 [00:01<00:00, 34.14it/s]
    
    Selected Keyframes at Frame 1: [0, 1]
    Mapping Time Step: 1: 100%|##########| 60/60 [00:02<00:00, 24.40it/s]
    3/10 frames received
    Tracking Time Step: 2: 100%|##########| 60/60 [00:01<00:00, 35.06it/s]
    
    Selected Keyframes at Frame 2: [0, 1, 2]
    Mapping Time Step: 2: 100%|##########| 60/60 [00:02<00:00, 22.20it/s]
    4/10 frames received
    Tracking Time Step: 3: 100%|##########| 60/60 [00:02<00:00, 27.32it/s]
    
    Selected Keyframes at Frame 3: [0, 1, 3]
    Mapping Time Step: 3: 100%|##########| 60/60 [00:02<00:00, 21.30it/s]
    5/10 frames received
    Tracking Time Step: 4: 100%|##########| 60/60 [00:02<00:00, 25.19it/s]
    
    Selected Keyframes at Frame 4: [1, 0, 3, 4]
    Mapping Time Step: 4: 100%|##########| 60/60 [00:02<00:00, 21.24it/s]
    6/10 frames received
    Tracking Time Step: 5: 100%|##########| 60/60 [00:02<00:00, 22.44it/s]
    
    Selected Keyframes at Frame 5: [0, 1, 3, 5]
    Mapping Time Step: 5: 100%|##########| 60/60 [00:02<00:00, 20.46it/s]
    7/10 frames received
    Tracking Time Step: 6: 100%|##########| 60/60 [00:02<00:00, 23.56it/s]
    
    Selected Keyframes at Frame 6: [1, 0, 3, 5, 6]
    Mapping Time Step: 6: 100%|##########| 60/60 [00:02<00:00, 20.36it/s]
    8/10 frames received
    Tracking Time Step: 7: 100%|##########| 60/60 [00:02<00:00, 21.45it/s]
    
    Selected Keyframes at Frame 7: [3, 1, 0, 5, 7]
    Mapping Time Step: 7: 100%|##########| 60/60 [00:02<00:00, 20.26it/s]
    9/10 frames received
    Tracking Time Step: 8: 100%|##########| 60/60 [00:02<00:00, 22.64it/s]
    
    Selected Keyframes at Frame 8: [3, 0, 1, 5, 7, 8]
    Mapping Time Step: 8: 100%|##########| 60/60 [00:02<00:00, 20.06it/s]
    10/10 frames received
    Tracking Time Step: 9: 100%|##########| 60/60 [00:02<00:00, 23.27it/s]
    
    Selected Keyframes at Frame 9: [5, 0, 1, 3, 7, 8, 9]
    Mapping Time Step: 9: 100%|##########| 60/60 [00:03<00:00, 18.10it/s]
    
    Average Tracking/Iteration Time: 39.425446369029856 ms
    Average Tracking/Frame Time: 2.129827785491943 s
    Average Mapping/Iteration Time: 47.29628523190816 ms
    Average Mapping/Frame Time: 2.838191199302673 s
    Saving parameters to: ././experiments/iPhone_Captures/splatam_demo\SplaTAM_iPhone
    Saved SplaTAM Splat to:  ././experiments/iPhone_Captures/splatam_demo\SplaTAM_iPhone
    (splatam)
    ```
    </details>

9. Visualize the result by running:
```
Nirmal@LAPTOP-HEKCV3H4 MINGW64 ~/Documents/SplaTAM (main)
$ python viz_scripts/final_recon.py configs/iphone/online_demo.py
```

Antony-htic, 2023-12-18:
While I was trying to perform online demo in windows 10 this error occurs. **Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-based projects this error still persists even after installing cuda**

IsaacBerman, 2023-12-19:
> While I was trying to perform online demo in windows 10 this error occurs. **Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-based projects this error still persists even after installing cuda**

I received the same error

bill9109, 2023-12-20:
> While I was trying to perform online demo in windows 10 this error occurs. **Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-based projects this error still persists even after installing cuda**

same

Nik-V9, 2023-12-26:
Hi, for help with the `diff-gaussian-rasterization` compilation on Windows, I would recommend referring to the 3DGS repo: https://github.com/graphdeco-inria/gaussian-splatting?tab=readme-ov-file#faq

Also, I think their closed issues have additional helpful info for Windows.

Nik-V9, 2023-12-26:
Potentially helpful for building environment on Windows: https://github.com/spla-tam/SplaTAM/issues/27#issuecomment-1857875316

darkstux1, 2024-01-02:
(splatam) C:\Users\gnbvr\SplaTAM>**python scripts/gaussian_splatting.py configs/iphone/gaussian_splatting.py**
System Paths:
C:\Users\gnbvr\SplaTAM
C:\Users\gnbvr\SplaTAM\scripts
C:\Users\gnbvr\pinokio\bin\miniconda\envs\splatam\python310.zip
C:\Users\gnbvr\pinokio\bin\miniconda\envs\splatam\DLLs
C:\Users\gnbvr\pinokio\bin\miniconda\envs\splatam\lib
C:\Users\gnbvr\pinokio\bin\miniconda\envs\splatam
C:\Users\gnbvr\pinokio\bin\miniconda\envs\splatam\lib\site-packages
C:\Users\gnbvr\pinokio\bin\miniconda\envs\splatam\lib\site-packages\win32
C:\Users\gnbvr\pinokio\bin\miniconda\envs\splatam\lib\site-packages\win32\lib
C:\Users\gnbvr\pinokio\bin\miniconda\envs\splatam\lib\site-packages\Pythonwin
C:\Users\gnbvr\pinokio\bin\miniconda\envs\splatam\lib\site-packages\torchaudio\backend\utils.py:62: UserWarning: No audio backend is available.
  warnings.warn("No audio backend is available.")
Seed set to: 0 (type: <class 'int'>)
Loaded Config:
{'workdir': '././experiments/iPhone_Captures/offline_demo', 'run_name': 'SplaTAM_iPhone', 'overwrite': False, 'depth_scale': 10.0, 'num_frames': 10, 'seed': 0, 'primary_device': 'cuda:0', 'map_every': 1, 'keyframe_every': 2, 'mapping_window_size': 32, 'report_global_progress_every': 100, 'eval_every': 1, 'scene_radius_depth_ratio': 3, 'mean_sq_dist_method': 'projective', 'report_iter_progress': False, 'load_checkpoint': False, 'checkpoint_time_idx': 130, 'save_checkpoints': False, 'checkpoint_interval': 5, 'use_wandb': False, 'data': {'dataset_name': 'nerfcapture', 'basedir': './experiments/iPhone_Captures', 'sequence': 'offline_demo', 'desired_image_height_init': 720, 'desired_image_width_init': 960, 'desired_image_height': 720, 'desired_image_width': 960, 'densification_image_height': 360, 'densification_image_width': 480, 'start': 0, 'end': -1, 'stride': 1, 'num_frames': 10}, 'tracking': {'use_gt_poses': False, 'forward_prop': True, 'visualize_tracking_loss': False, 'num_iters': 60, 'use_sil_for_loss': True, 'sil_thres': 0.99, 'use_l1': True, 'use_depth_loss_thres': True, 'depth_loss_thres': 20000, 'ignore_outlier_depth_loss': False, 'use_uncertainty_for_loss_mask': False, 'use_uncertainty_for_loss': False, 'use_chamfer': False, 'loss_weights': {'im': 0.5, 'depth': 1.0}, 'lrs': {'means3D': 0.0, 'rgb_colors': 0.0, 'unnorm_rotations': 0.0, 'logit_opacities': 0.0, 'log_scales': 0.0, 'cam_unnorm_rots': 0.001, 'cam_trans': 0.004}}, 'mapping': {'num_iters': 60, 'add_new_gaussians': True, 'sil_thres': 0.5, 'use_l1': True, 'ignore_outlier_depth_loss': False, 'use_sil_for_loss': False, 'use_uncertainty_for_loss_mask': False, 'use_uncertainty_for_loss': False, 'use_chamfer': False, 'loss_weights': {'im': 0.5, 'depth': 1.0}, 'lrs': {'means3D': 0.0001, 'rgb_colors': 0.0025, 'unnorm_rotations': 0.001, 'logit_opacities': 0.05, 'log_scales': 0.001, 'cam_unnorm_rots': 0.0, 'cam_trans': 0.0}, 'prune_gaussians': True, 'pruning_dict': {'start_after': 0, 'remove_big_after': 0, 'stop_after': 20, 'prune_every': 20, 'removal_opacity_threshold': 0.005, 'final_removal_opacity_threshold': 0.005, 'reset_opacities': False, 'reset_opacities_every': 500}, 'use_gaussian_splatting_densification': False, 'densify_dict': {'start_after': 500, 'remove_big_after': 3000, 'stop_after': 5000, 'densify_every': 100, 'grad_thresh': 0.0002, 'num_to_split_into': 2, 'removal_opacity_threshold': 0.005, 'final_removal_opacity_threshold': 0.005, 'reset_opacities_every': 3000}}, 'viz': {'render_mode': 'color', 'offset_first_viz_cam': True, 'show_sil': False, 'visualize_cams': True, 'viz_w': 600, 'viz_h': 340, 'viz_near': 0.01, 'viz_far': 100.0, 'view_scale': 2, 'viz_fps': 5, 'enter_interactive_post_online': False}}
Loading Dataset ...
Traceback (most recent call last):
  File "C:\Users\gnbvr\SplaTAM\scripts\gaussian_splatting.py", line 626, in <module>
    rgbd_slam(experiment.config)
  File "C:\Users\gnbvr\SplaTAM\scripts\gaussian_splatting.py", line 366, in rgbd_slam
    dataset = get_dataset(
  File "C:\Users\gnbvr\SplaTAM\scripts\gaussian_splatting.py", line 71, in get_dataset
    return NeRFCaptureDataset(basedir, sequence, **kwargs)
  File "C:\Users\gnbvr\SplaTAM\datasets\gradslam_datasets\nerfcapture.py", line 39, in __init__
    self.cams_metadata = self.load_cams_metadata()
  File "C:\Users\gnbvr\SplaTAM\datasets\gradslam_datasets\nerfcapture.py", line 72, in load_cams_metadata
    cams_metadata = json.load(open(cams_metadata_path, "r"))
**FileNotFoundError: [Errno 2] No such file or directory: './experiments/iPhone_Captures\\offline_demo/transforms.json'**

I shoot with my iPhone 14 in online mode, but I cannot get training for Gaussian splatting.

darkstux1, 2024-02-23:
> The same too, and I don't know how to fix it.

05355774619
Seyit kemal GÜNGÖR
I can help you if you contact me here

Shanzhaguoo ***@***.***>, 23 Şub 2024 Cum, 11:04 tarihinde
şunu yazdı:

jledner, 2024-03-16:
running into the following issue. any suggestions?

`$ pip install -r requirements.txt
Collecting git+https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth.git@cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 (from -r requirements.txt (line 16))
  Cloning https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth.git (to revision cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110) to c:\users\jakeledner\appdata\local\temp\pip-req-build-c6qm11vt
  Running command git clone --filter=blob:none --quiet https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth.git 'C:\Users\JakeLedner\AppData\Local\Temp\pip-req-build-c6qm11vt'
  Running command git rev-parse -q --verify 'sha^cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110'
  Running command git fetch -q https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth.git cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110
  Resolved https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth.git to commit cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110
  Running command git submodule update --init --recursive -q
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'error'
  error: subprocess-exited-with-error

  python setup.py egg_info did not run successfully.
  exit code: 1

  [12 lines of output]
  Traceback (most recent call last):
    File "<string>", line 2, in <module>
    File "<pip-setuptools-caller>", line 34, in <module>
    File "C:\Users\JakeLedner\AppData\Local\Temp\pip-req-build-c6qm11vt\setup.py", line 21, in <module>
      CUDAExtension(
    File "C:\Users\JakeLedner\anaconda3\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 983, in CUDAExtension
      library_dirs += library_paths(cuda=True)
    File "C:\Users\JakeLedner\anaconda3\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 1105, in library_paths
      paths.append(_join_cuda_home(lib_dir))
    File "C:\Users\JakeLedner\anaconda3\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 2125, in _join_cuda_home
      raise EnvironmentError('CUDA_HOME environment variable is not set. '
  OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.
  [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

Encountered error while generating package metadata.

See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.
`

Xiaobizaizi11, 2024-03-31:
The above exception was the direct cause of the following exception:

      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\86186\AppData\Local\Temp\pip-req-build-hrwyok9k\setup.py", line 17, in <module>
          setup(
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\core.py", line 185, in setup
          return run_commands(dist)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\core.py", line 201, in run_commands
          dist.run_commands()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\dist.py", line 989, in run_command
          super().run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\wheel\bdist_wheel.py", line 364, in run
          self.run_command("build")
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\dist.py", line 989, in run_command
          super().run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build.py", line 131, in run
          self.run_command(cmd_name)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\dist.py", line 989, in run_command
          super().run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\command\build_ext.py", line 88, in run
          _build_ext.run(self)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 345, in run
          self.build_extensions()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 765, in build_extensions
          build_ext.build_extensions(self)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 467, in build_extensions
          self._build_extensions_serial()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 493, in _build_extensions_serial
          self.build_extension(ext)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\command\build_ext.py", line 249, in build_extension
          _build_ext.build_extension(self, ext)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 548, in build_extension
          objects = self.compiler.compile(
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 738, in win_wrap_ninja_compile
          _write_ninja_file_and_compile_objects(
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 1487, in _write_ninja_file_and_compile_objects
          _run_ninja_build(
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 1824, in _run_ninja_build
          raise RuntimeError(message) from e
      RuntimeError: Error compiling objects for extension
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for diff-gaussian-rasterization
  Running setup.py clean for diff-gaussian-rasterization
Failed to build diff-gaussian-rasterization
ERROR: Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-

How should I solve the error I encountered when executing pip install -r requirements.txt on Windows?

GaigeY, 2024-06-21:
> The above exception was the direct cause of the following exception:
>
> ```
>   Traceback (most recent call last):
>     File "<string>", line 2, in <module>
>     File "<pip-setuptools-caller>", line 34, [...]

Hi @Xiaobizaizi11 , I also met this problem, and I located the problem via cmake. Maybe you should check your cmake result.