spla-tam/SplaTAM #11: Setting up conda environment takes forever

nekoshadow1, 2023-12-08:
First of all, thank you for this fantastic work! I really like to test it out with my own data. But it takes forever to set up the environment on my workbench. 

This is the detailed setting of my workbench:
Platform: GCP
Hardware: 1 NVIDIA L4 GPU, 4 vCPUs, 16GB RAM
CUDA: 11.3 or 11.8

I noticed that the environment.yml said the code requires 'cudatoolkit=11.6'. I wonder if this is why it is stuck in setting up the environment.

Any suggestion is appreciated!!

JayKarhade, 2023-12-08:
Hi @nekoshadow1 , yes we also observed the slow conda installation. 

I just updated the installation instructions  and it should be faster now. clCould you try that?

nekoshadow1, 2023-12-08:
> Hi @nekoshadow1 , yes we also observed the slow conda installation.
> 
> I just updated the installation instructions and it should be faster now. clCould you try that?

Thanks for the lightning-fast fix! Yes, this time the installation is much faster and successful. I will close this issue.

closed 2023-12-08 by nekoshadow1