spla-tam/SplaTAM #13: Installation error during `pip install -r requirements.txt`

nirmalsnair, 2023-12-08:
🤗 Thanks for making the code publicly available.

I used the following instructions to set up the environment on my Windows 10 laptop. (see full installation log below)

```python
conda create -n splatam python=3.10
conda activate splatam
conda install -c "nvidia/label/cuda-11.6.0" cuda-toolkit
conda install pytorch==1.12.1 torchvision==0.13.1 torchaudio==0.12.1 cudatoolkit=11.6 -c pytorch -c conda-forge
pip install -r requirements.txt
```

But at the final step of the installation, `pip` shows the following error.

```python
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
tea-console 0.0.6 requires rich~=9.11.0, but you have rich 13.7.0 which is incompatible.
typer 0.3.2 requires click<7.2.0,>=7.1.1, but you have click 8.1.7 which is incompatible.
Successfully installed Click-8.1.7 Flask-3.0.0 GitPython-3.1.40 Jinja2-3.1.2 MarkupSafe-2.1.3 Werkzeug-3.0.1 ansi2html-1.8.0 appdirs-1.4.4 asttokens-2.4.1 attrs-23.1.0 blinker-1.7.0 colorama-0.4.6 comm-0.2.0 configargparse-1.7 contourpy-1.2.0 cycler-0.12.1 cyclonedds-0.10.2 dash-2.14.2 dash-core-components-2.0.0 dash-html-components-2.0.0 dash-table-5.0.0 decorator-5.1.1 diff-gaussian-rasterization-0.0.0 docker-pycreds-0.4.0 exceptiongroup-1.2.0 executing-2.0.1 fastjsonschema-2.19.0 fonttools-4.46.0 gitdb-4.0.11 imageio-2.33.0 importlib-metadata-7.0.0 ipython-8.18.1 ipywidgets-8.1.1 itsdangerous-2.1.2 jedi-0.19.1 jsonschema-4.20.0 jsonschema-specifications-2023.11.2 jupyter_core-5.5.0 jupyterlab-widgets-3.0.9 kiwisolver-1.4.5 kornia-0.7.0 lightning-utilities-0.10.0 lpips-0.1.4 markdown-it-py-3.0.0 matplotlib-3.8.2 matplotlib-inline-0.1.6 mdurl-0.1.2 natsort-8.4.0 nbformat-5.5.0 nest-asyncio-1.5.8 open3d-0.16.0 opencv-python-4.8.1.78 packaging-23.2 parso-0.8.3 platformdirs-4.1.0 plotly-5.18.0 prompt-toolkit-3.0.41 protobuf-4.25.1 pure-eval-0.2.2 pygments-2.17.2 pyparsing-3.1.1 python-dateutil-2.8.2 pytorch-msssim-1.0.0 pywin32-306 pyyaml-6.0.1 referencing-0.32.0 retrying-1.3.4 rich-13.7.0 rich-click-1.7.2 rpds-py-0.13.2 scipy-1.11.4 sentry-sdk-1.38.0 setproctitle-1.3.3 six-1.16.0 smmap-5.0.1 stack-data-0.6.3 tenacity-8.2.3 torchmetrics-1.2.1 tqdm-4.65.0 traitlets-5.14.0 typing-extensions-4.8.0 wandb-0.16.1 wcwidth-0.2.12 widgetsnbextension-4.0.9 zipp-3.17.0
```

I'm not sure if the installation was successful, because on running:
`(splatam) C:\Users\Nirmal\Documents\SplaTAM>bash bash_scripts/online_demo.bash configs/iphone/online_demo.py`

I get the following error:

```python
<3>WSL (8) ERROR: CreateProcessParseCommon:754: getpwuid(0) failed 2
Processing fstab with mount -a failed.

<3>WSL (8) ERROR: CreateProcessEntryCommon:331: getpwuid(0) failed 2
<3>WSL (8) ERROR: CreateProcessEntryCommon:502: execvpe /bin/bash failed 2
<3>WSL (8) ERROR: CreateProcessEntryCommon:505: Create process not expected to return
```

---

<details>
<summary>View the full installation log</summary>

```python
(base) C:\WINDOWS\system32>conda env list
# conda environments:
#
base                  *  C:\Anaconda3
voldor                   C:\Users\Nirmal\.conda\envs\voldor
work                     C:\Users\Nirmal\.conda\envs\work
```

```
(base) C:\WINDOWS\system32>conda create -n splatam python=3.10
WARNING: A directory already exists at the target location 'C:\Anaconda3\envs\splatam'
but it is not a conda environment.
Continue creating environment (y/[n])? y

Channels:
 - defaults
Platform: win-64
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: C:\Anaconda3\envs\splatam

  added / updated specs:
    - python=3.10

The following NEW packages will be INSTALLED:

[... 517 lines cut ...]
Downloading wcwidth-0.2.12-py2.py3-none-any.whl (34 kB)
Building wheels for collected packages: diff-gaussian-rasterization
  Building wheel for diff-gaussian-rasterization (setup.py) ... done
  Created wheel for diff-gaussian-rasterization: filename=diff_gaussian_rasterization-0.0.0-cp310-cp310-win_amd64.whl size=323784 sha256=c60080785f3de0f801684ae41ddb0359833d3c1eb37cb67b304e48b99497a16f
  Stored in directory: c:\users\nirmal\appdata\local\pip\cache\wheels\7f\06\a4\a99846e65c144ce0493fdc05cecd76e841ecadd5385d3db393
Successfully built diff-gaussian-rasterization
Installing collected packages: wcwidth, pywin32, pure-eval, fastjsonschema, diff-gaussian-rasterization, dash-table, dash-html-components, dash-core-components, appdirs, zipp, widgetsnbextension, typing-extensions, traitlets, tenacity, smmap, six, setproctitle, sentry-sdk, scipy, rpds-py, pyyaml, pyparsing, pygments, protobuf, prompt-toolkit, platformdirs, parso, packaging, opencv-python, nest-asyncio, natsort, mdurl, MarkupSafe, kiwisolver, jupyterlab-widgets, itsdangerous, imageio, fonttools, executing, exceptiongroup, decorator, cycler, contourpy, configargparse, colorama, blinker, attrs, ansi2html, Werkzeug, tqdm, retrying, referencing, python-dateutil, plotly, matplotlib-inline, markdown-it-py, lightning-utilities, jupyter_core, Jinja2, jedi, importlib-metadata, gitdb, docker-pycreds, comm, Click, asttokens, torchmetrics, stack-data, rich, pytorch-msssim, matplotlib, kornia, jsonschema-specifications, GitPython, Flask, wandb, rich-click, lpips, jsonschema, ipython, dash, nbformat, ipywidgets, cyclonedds, open3d
  Attempting uninstall: typing-extensions
    Found existing installation: typing-extensions 3.10.0.2
    Uninstalling typing-extensions-3.10.0.2:
      Successfully uninstalled typing-extensions-3.10.0.2
  Attempting uninstall: Click
    Found existing installation: click 7.1.2
    Uninstalling click-7.1.2:
      Successfully uninstalled click-7.1.2
  Attempting uninstall: rich
    Found existing installation: rich 9.11.1
    Uninstalling rich-9.11.1:
      Successfully uninstalled rich-9.11.1
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
tea-console 0.0.6 requires rich~=9.11.0, but you have rich 13.7.0 which is incompatible.
typer 0.3.2 requires click<7.2.0,>=7.1.1, but you have click 8.1.7 which is incompatible.
Successfully installed Click-8.1.7 Flask-3.0.0 GitPython-3.1.40 Jinja2-3.1.2 MarkupSafe-2.1.3 Werkzeug-3.0.1 ansi2html-1.8.0 appdirs-1.4.4 asttokens-2.4.1 attrs-23.1.0 blinker-1.7.0 colorama-0.4.6 comm-0.2.0 configargparse-1.7 contourpy-1.2.0 cycler-0.12.1 cyclonedds-0.10.2 dash-2.14.2 dash-core-components-2.0.0 dash-html-components-2.0.0 dash-table-5.0.0 decorator-5.1.1 diff-gaussian-rasterization-0.0.0 docker-pycreds-0.4.0 exceptiongroup-1.2.0 executing-2.0.1 fastjsonschema-2.19.0 fonttools-4.46.0 gitdb-4.0.11 imageio-2.33.0 importlib-metadata-7.0.0 ipython-8.18.1 ipywidgets-8.1.1 itsdangerous-2.1.2 jedi-0.19.1 jsonschema-4.20.0 jsonschema-specifications-2023.11.2 jupyter_core-5.5.0 jupyterlab-widgets-3.0.9 kiwisolver-1.4.5 kornia-0.7.0 lightning-utilities-0.10.0 lpips-0.1.4 markdown-it-py-3.0.0 matplotlib-3.8.2 matplotlib-inline-0.1.6 mdurl-0.1.2 natsort-8.4.0 nbformat-5.5.0 nest-asyncio-1.5.8 open3d-0.16.0 opencv-python-4.8.1.78 packaging-23.2 parso-0.8.3 platformdirs-4.1.0 plotly-5.18.0 prompt-toolkit-3.0.41 protobuf-4.25.1 pure-eval-0.2.2 pygments-2.17.2 pyparsing-3.1.1 python-dateutil-2.8.2 pytorch-msssim-1.0.0 pywin32-306 pyyaml-6.0.1 referencing-0.32.0 retrying-1.3.4 rich-13.7.0 rich-click-1.7.2 rpds-py-0.13.2 scipy-1.11.4 sentry-sdk-1.38.0 setproctitle-1.3.3 six-1.16.0 smmap-5.0.1 stack-data-0.6.3 tenacity-8.2.3 torchmetrics-1.2.1 tqdm-4.65.0 traitlets-5.14.0 typing-extensions-4.8.0 wandb-0.16.1 wcwidth-0.2.12 widgetsnbextension-4.0.9 zipp-3.17.0
```
</details>

JayKarhade, 2023-12-08:
Hi @nirmalsnair we did not test our code on windows. 

Could you first confirm if the replica or other benchmark runs successfully?

nirmalsnair, 2023-12-09:
The error I got on running `online_demo.py` was not due to SplaTAM, but because I was running it on Anaconda Prompt on Windows. Running the code on Git Bash solved the issue.

I've posted the full details [here](https://github.com/spla-tam/SplaTAM/issues/9#issuecomment-1848348403). Thanks!

closed 2023-12-09 by nirmalsnair