spla-tam/SplaTAM #15: Git clone with --recursive

java311, 2023-12-08:
Hello 

I think I found an issue related to the submodules present in this repository. 

**Issue.** 
- `git clone` with `--recursive` option fails
**Possible fix.** 
- I think the issue is cause by setting the repository using SSH. The issue solves when I change the repository to HTTPS.
By editing `.gitmodules` and then replace line:
`url = git@github.com:JonathonLuiten/diff-gaussian-rasterization-w-depth.git`
to:
`url = https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth.git`

Then I reload the submodules using : `git submodule update --init --recursive`

But even so, I still issues with the submodules, during conda env install. 😑

darshit-desai, 2023-12-09:
Hi @java311 just making sure are you ssh'd to your github account from the terminal you're using in windows. I think in my case it asked me the ssh password when I used the `git submodule update --init --recursive` and it was able to clone it. This looks like more of a git issue rather then a repo issue

Nik-V9, 2023-12-26:
Closing this issue due to inactivity, please feel free to re-open. I would recommend following: https://github.com/spla-tam/SplaTAM/issues/9

closed 2023-12-26 by Nik-V9