spla-tam/SplaTAM #20: Visualisation of SLAM on benchmarking datasets

darshit-desai, 2023-12-10:
Hi Authors, Is there a way to visualize the reconstruction while the slam runs on the benchmarking dataset. I just see a command for running the splatam.py file `python scripts/splatam.py configs/replica/splatam.py` in case of replica dataset

JayKarhade, 2023-12-10:
Hi @darshit-desai ,

Currently we've provided support for visualization only after the slam is completed. This is through the viz_scripts/ files, and have instructions in the Readme.

However, we do not yet have support for viz while the demo optimization is happening. That is currently in the works. 

See https://github.com/spla-tam/SplaTAM/issues/8#issuecomment-1845351121

JayKarhade, 2023-12-13:
Closing this for now since we have this as TODO. Feel free to reopen if something is unanswered.

closed 2023-12-13 by JayKarhade