spla-tam/SplaTAM #25: slow evaluation speed

FrontierBreaker, 2023-12-12:
Hi, I find that the training time is very fast but it seems to be stuck during the final evaluation step. Could you please tell me why? Thanks!
Evaluating Final Parameters ...
  7%|██▋                                   | 139/2000 [42:48<7:12:19, 13.94s/it]

JayKarhade, 2023-12-12:
Is it getting stuck at 139 step or is it slow inference?

The evaluation is slower because we calculate LPIPS and also store images.

FrontierBreaker, 2023-12-12:
The inference is slow. Is the inference speed normal?

Nik-V9, 2023-12-12:
Hi, no, this speed is not normal. Generally, the final evaluation is slow since we write the final output for each eval frame to the disk. However, it shouldn't be 7 hr/sequence. This might be an issue with storage on your end?

FrontierBreaker, 2023-12-12:
okay, I will check it. Thank you!

closed 2023-12-12 by FrontierBreaker