spla-tam/SplaTAM #27: Error when creating env from environment.yml

LukasPoque, 2023-12-12:
Sorry, if I just made a stupid mistake, I'm not so familiar with anaconda, but I tried to create the environment from your blueprint with `conda env create -f environment.yml` and after a while, this error shows up:
```
Installing pip dependencies: - Ran pip subprocess with arguments:
['C:\\Users\\Lukas\\anaconda3\\envs\\splatam\\python.exe', '-m', 'pip', 'install', '-U', '-r', 'C:\\Users\\Lukas\\Code\\external\\SplaTAM\\condaenv.h3hrhgos.requirements.txt', '--exists-action=b']
Pip subprocess output:
Collecting git+https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 (from -r C:\Users\Lukas\Code\external\SplaTAM\condaenv.h3hrhgos.requirements.txt (line 5))
  Cloning https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 to c:\users\lukas\appdata\local\temp\pip-req-build-t9ur40ad

Pip subprocess error:
  Running command git clone --filter=blob:none --quiet https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 'C:\Users\Lukas\AppData\Local\Temp\pip-req-build-t9ur40ad'
  fatal: repository 'https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110/' not found
  error: subprocess-exited-with-error

  × git clone --filter=blob:none --quiet https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 'C:\Users\Lukas\AppData\Local\Temp\pip-req-build-t9ur40ad' did not run successfully.
  │ exit code: 128
  ╰─> See above for output.

```

Do you have any idea why the repo could not be found? I can see the repository if I copy the link to my browser.

If it matters, I'm testing this on windows.

nirmalsnair, 2023-12-12:
I too got the same error (Windows 10). Running the following (recommended method) worked for me.

```
conda create -n splatam python=3.10
conda activate splatam
conda install -c "nvidia/label/cuda-11.6.0" cuda-toolkit
conda install pytorch==1.12.1 torchvision==0.13.1 torchaudio==0.12.1 cudatoolkit=11.6 -c pytorch -c conda-forge
pip install -r requirements.txt
```

LukasPoque, 2023-12-12:
Sadly I'm getting another error while trying this:
```
Building wheels for collected packages: diff-gaussian-rasterization
  Building wheel for diff-gaussian-rasterization (setup.py) ... error
  error: subprocess-exited-with-error

  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [57 lines of output]
      running bdist_wheel
      C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py:411: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
        warnings.warn(msg.format('we could not find ninja.'))
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-cpython-310
      creating build\lib.win-amd64-cpython-310\diff_gaussian_rasterization
      copying diff_gaussian_rasterization\__init__.py -> build\lib.win-amd64-cpython-310\diff_gaussian_rasterization
      running build_ext
      C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py:346: UserWarning: Error checking compiler version for cl: [WinError 2] The system cannot find the file specified
        warnings.warn(f'Error checking compiler version for {compiler}: {error}')
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\Lukas\AppData\Local\Temp\pip-req-build-j4r3iu10\setup.py", line 17, in <module>
          setup(
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\__init__.py", line 107, in setup
          return distutils.core.setup(**attrs)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\core.py", line 185, in setup
          return run_commands(dist)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\core.py", line 201, in run_commands
          dist.run_commands()
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\dist.py", line 1234, in run_command
          super().run_command(command)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\wheel\bdist_wheel.py", line 364, in run
          self.run_command("build")
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\dist.py", line 1234, in run_command
          super().run_command(command)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\command\build.py", line 131, in run
          self.run_command(cmd_name)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\dist.py", line 1234, in run_command
          super().run_command(command)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\command\build_ext.py", line 84, in run
          _build_ext.run(self)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 345, in run
          self.build_extensions()
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 434, in build_extensions
          self._check_cuda_version(compiler_name, compiler_version)
        File "C:\Users\Lukas\anaconda3\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 812, in _check_cuda_version
          raise RuntimeError(CUDA_MISMATCH_MESSAGE.format(cuda_str_version, torch.version.cuda))
      RuntimeError:
      The detected CUDA version (12.3) mismatches the version that was used to compile
      PyTorch (11.6). Please make sure to use the same CUDA versions.

      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for diff-gaussian-rasterization
  Running setup.py clean for diff-gaussian-rasterization
Failed to build diff-gaussian-rasterization
ERROR: Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-based projects
```

I think this is strange because `conda install -c "nvidia/label/cuda-11.6.0" cuda-toolkit` should install the correct cuda version or am I wrong?

Nik-V9, 2023-12-12:
Hi, Yes, this seems strange. Can you try doing `nvcc --version`?

pablovela5620, 2023-12-14:
I am also having issues, this is my error

when I run nvcc --version I get 
```
splatam) pablo@pablo-ubuntu:~/0Dev/repos/SplaTAM$ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Fri_Dec_17_18:16:03_PST_2021
Cuda compilation tools, release 11.6, V11.6.55
Build cuda_11.6.r11.6/compiler.30794723_0
(splatam) pablo@pablo-ubuntu:~/0Dev/repos/Sp
```

```
Building wheels for collected packages: diff-gaussian-rasterization
  Building wheel for diff-gaussian-rasterization (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [115 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-310
      creating build/lib.linux-x86_64-cpython-310/diff_gaussian_rasterization
      copying diff_gaussian_rasterization/__init__.py -> build/lib.linux-x86_64-cpython-310/diff_gaussian_rasterization
      running build_ext
      /home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py:820: UserWarning: There are no g++ version bounds defined for CUDA version 11.6
        warnings.warn(f'There are no {compiler_name} version bounds defined for CUDA version {cuda_str_version}')
      building 'diff_gaussian_rasterization._C' extension
      creating /tmp/pip-req-build-jbtis6yf/build/temp.linux-x86_64-cpython-310
      creating /tmp/pip-req-build-jbtis6yf/build/temp.linux-x86_64-cpython-310/cuda_rasterizer
      Emitting ninja build file /tmp/pip-req-build-jbtis6yf/build/temp.linux-x86_64-cpython-310/build.ninja...
      Compiling objects...
      Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
      [1/5] /home/pablo/micromamba/envs/splatam/bin/nvcc  -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include/TH -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include/THC -I/home/pablo/micromamba/envs/splatam/include -I/home/pablo/micromamba/envs/splatam/include/python3.10 -c -c /tmp/pip-req-build-jbtis6yf/cuda_rasterizer/forward.cu -o /tmp/pip-req-build-jbtis6yf/build/temp.linux-x86_64-cpython-310/cuda_rasterizer/forward.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -I/tmp/pip-req-build-jbtis6yf/third_party/glm/ -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_61,code=compute_61 -gencode=arch=compute_61,code=sm_61 -std=c++14
      /tmp/pip-req-build-jbtis6yf/cuda_rasterizer/auxiliary.h(151): warning #177-D: variable "p_proj" was declared but never referenced
      
      [2/5] /home/pablo/micromamba/envs/splatam/bin/nvcc  -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include/TH -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include/THC -I/home/pablo/micromamba/envs/splatam/include -I/home/pablo/micromamba/envs/splatam/include/python3.10 -c -c /tmp/pip-req-build-jbtis6yf/cuda_rasterizer/backward.cu -o /tmp/pip-req-build-jbtis6yf/build/temp.linux-x86_64-cpython-310/cuda_rasterizer/backward.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -I/tmp/pip-req-build-jbtis6yf/third_party/glm/ -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_61,code=compute_61 -gencode=arch=compute_61,code=sm_61 -std=c++14
      /tmp/pip-req-build-jbtis6yf/cuda_rasterizer/auxiliary.h(151): warning #177-D: variable "p_proj" was declared but never referenced
      
      [3/5] /home/pablo/micromamba/envs/splatam/bin/nvcc  -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include/torch/csrc/api/include -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include/TH -I/home/pablo/micromamba/envs/splatam/lib/python3.10/site-packages/torch/include/THC -I/home/pablo/micromamba/envs/splatam/include -I/home/pablo/micromamba/envs/splatam/include/python3.10 -c -c /tmp/pip-req-build-jbtis6yf/cuda_rasterizer/rasterizer_impl.cu -o /tmp/pip-req-build-jbtis6yf/build/temp.linux-x86_64-cpython-310/cuda_rasterizer/rasterizer_impl.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -I/tmp/pip-req-build-jbtis6yf/third_party/glm/ -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' [... cut ...]
          raise RuntimeError(message) from e
      RuntimeError: Error compiling objects for extension
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for diff-gaussian-rasterization
  Running setup.py clean for diff-gaussian-rasterization
Failed to build diff-gaussian-rasterization
ERROR: Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-based projects
```

Nik-V9, 2023-12-26:
Thanks for this discussion! Unfortunately, we have limited support on Windows.

Please follow this thread for further updates: https://github.com/spla-tam/SplaTAM/issues/9

closed 2023-12-26 by Nik-V9