spla-tam/SplaTAM #30: a small question about ref_w2c

lvmingzhe, 2023-12-13:
Hi, thanks for your great work.
I`m trying to understand this line, but I have no idea what is in the ```curr_gt_w2c[-1]```
https://github.com/spla-tam/SplaTAM/blob/bbaf5cc5754bf1034b33902007872c694e412a31/scripts/splatam.py#L741

could you please explain it?

JayKarhade, 2023-12-13:
Hi @lvmingzhe , that is basically getting the latest  pose w.r.t to the first frame(which we assume as origin). This is triggered only when you use the gt_poses for tracking. 

You can find curr_gt_w2c defined here : 
https://github.com/spla-tam/SplaTAM/blob/bbaf5cc5754bf1034b33902007872c694e412a31/scripts/splatam.py#L645C18-L645C18

lvmingzhe, 2023-12-13:
thx, I got it is to get the current ground truth pose relative to frame 0.

closed 2023-12-13 by lvmingzhe