spla-tam/SplaTAM #31: How to use the Nerfcapture offline dataset and about the usage of ground truth.

liaiwanovo, 2023-12-14:
Thank you and your team for the excellent code. I have a question about using Nerfcapture to record a dataset offline. After extracting and sending the dataset to my computer and placing it in the ./data folder, as well as modifying all the configuration files, it keeps showing "waiting for frames". How can I solve this issue?Looking forward to your reply！
[image]

Nik-V9, 2023-12-26:
Hi, Thanks for your queries!
1. We use the ARKit poses as Ground Truth for the iPhone demo.
2. The coordinate system is the first frame (everything is relative to the first frame).

liaiwanovo, 2023-12-27:
Thank you for your response and excellent work！

closed 2023-12-27 by liaiwanovo