spla-tam/SplaTAM #33: How to get OBJ file from params.npz

hhcxx2006, 2023-12-15:
Hi, I am very excited about your work. I have used your algorithm to reconstruct my scene. I have get the `params.npz` file. Now I want to use your reconstructed scene model in our project. We need OBJ file format to import the model. We want to import the scene for our robot simulation system. Could you please tell me how to convert the `params.npz` file to OBJ file? Thanks very much!  It is just the point cloud. How to convert the point cloud to mesh?

Nik-V9, 2023-12-26:
Hi, we currently don't have a method to convert the final Splat to an OBJ file. There's recent work on converting Gaussian Splats to High-Fidelity Meshes! They also have code available. I would recommend trying that out (we haven't tried it yet): https://imagine.enpc.fr/~guedona/sugar/

Another option is to run Poisson Surface Reconstruction on the point cloud (centers of the 3D Gaussians).

YunnaChen, 2024-01-10:
> Hi, I am very excited about your work. I have used your algorithm to reconstruct my scene. I have get the `params.npz` file. Now I want to use your reconstructed scene model in our project. We need [...]

Hi，have you tried?

WN-Wolf, 2024-01-23:
> Hi, I am very excited about your work. I have used your algorithm to reconstruct my scene. I have get the `params.npz` file. Now I want to use your reconstructed scene model in our project. We need [...]

@hhcxx2006 Hello, I used D435 camera to create a dataset. But when I run myself dataset, there were many errors. 
In this command, "python scripts/splatam.py configs/my_data/splatam.py", I don't know how to modify "splatam.py" file or refer to which dataset's config file(Replica or TUM-RGBD has different config file). So, Could you please give me some suggestions or share your "splatam.py" file? 
Thanks a lot.

zhaosiyuan1098, 2024-03-20:
hi，I found a good repository coming recently, it may helps you to get a .ply file:[gaustudio](https://github.com/GAP-LAB-CUHK-SZ/gaustudio)