spla-tam/SplaTAM #36: Could not build wheels for diff-gaussian-rasterization

bill9109, 2023-12-18:
I have been encountering the 'Could not build wheels for diff-gaussian-rasterization' issue. I tried:

Installing in the normal order.
Installing according to environment.yml.
The result was the same in both cases.
So, I attempted to directly install from https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth by using setup.py . Here is the error message I encountered:

```
E:\anaconda3\envs\splatam\lib\site-packages\torch\include\pybind11\cast.h(1429): error: too few arguments for template template parameter 'Tuple'
detected during the instantiation of class 'pybind11::detail::tuple_caster<Tuple, Ts...> [with Tuple=std::pair, Ts=<T1, T2>]'
(1507): here

E:\anaconda3\envs\splatam\lib\site-packages\torch\include\pybind11\cast.h(1503): error: too few arguments for template template parameter 'Tuple'
detected during the instantiation of class 'pybind11::detail::tuple_caster<Tuple, Ts...> [with Tuple=std::pair, Ts=<T1, T2>]'
(1507): here

2 errors detected in the compilation of 'rasterize_points.cu'.
rasterize_points.cu
error: command 'E:\anaconda3\envs\splatam\bin\nvcc.exe' failed with exit code 1
``` 

How can I resolve this issue?

spirobel, 2023-12-19:
https://github.com/spla-tam/SplaTAM/pull/24 <--- solution in there

bill9109, 2023-12-20:
but I'm using a Windows, so maybe version problem with MSVC? VS2022, in my case.

bill9109, 2023-12-20:
well…… I don't know how but I might fix this by replace E:\anaconda3\envs\splatam\Lib\site-packages\torch\include\pybind11\cast.h by this one https://gist.github.com/johnbanq/f4e4e8ba9253ba0dc4e09ad0e0b73ebb

Nik-V9, 2023-12-26:
Hi, we currently don't have good support for Windows. We are working on this (@JayKarhade).

Please use this issue thread for further discussion: https://github.com/spla-tam/SplaTAM/issues/9

closed 2023-12-26 by Nik-V9

Xiaobizaizi11, 2024-03-31:
The above exception was the direct cause of the following exception:

      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "C:\Users\86186\AppData\Local\Temp\pip-req-build-hrwyok9k\setup.py", line 17, in <module>
          setup(
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\core.py", line 185, in setup
          return run_commands(dist)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\core.py", line 201, in run_commands
          dist.run_commands()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\dist.py", line 989, in run_command
          super().run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\wheel\bdist_wheel.py", line 364, in run
          self.run_command("build")
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\dist.py", line 989, in run_command
          super().run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build.py", line 131, in run
          self.run_command(cmd_name)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\dist.py", line 989, in run_command
          super().run_command(command)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\command\build_ext.py", line 88, in run
          _build_ext.run(self)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 345, in run
          self.build_extensions()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 765, in build_extensions
          build_ext.build_extensions(self)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 467, in build_extensions
          self._build_extensions_serial()
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 493, in _build_extensions_serial
          self.build_extension(ext)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\command\build_ext.py", line 249, in build_extension
          _build_ext.build_extension(self, ext)
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 548, in build_extension
          objects = self.compiler.compile(
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 738, in win_wrap_ninja_compile
          _write_ninja_file_and_compile_objects(
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 1487, in _write_ninja_file_and_compile_objects
          _run_ninja_build(
        File "D:\Deeplearning\anaconda\envs\splatam\lib\site-packages\torch\utils\cpp_extension.py", line 1824, in _run_ninja_build
          raise RuntimeError(message) from e
      RuntimeError: Error compiling objects for extension
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for diff-gaussian-rasterization
  Running setup.py clean for diff-gaussian-rasterization
Failed to build diff-gaussian-rasterization
ERROR: Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-

How should I solve the error I encountered when executing pip install -r requirements.txt on Windows?