spla-tam/SplaTAM #38: Bug when setting the config use_gaussian_splatting_densification=True

Ramseyous0109, 2023-12-19:
Hi, thanks for this brilliant work.
I came across a bug running an offline demo on the replica dataset. I found that in the config file, use gaussian splatting-based densification is set to false:

`use_gaussian_splatting_densification=False, # Use Gaussian Splatting-based Densification during Mapping`

When I set this config to `True`, a bug appeared when training: 

`File "/home/user/splatam/SplaTAM/utils/slam_external.py", line 101, in accumulate_mean2d_gradient    variables['means2D_gradient_accum'][variables['seen']] += torch.norm(IndexError: The shape of the mask [853079] at index 0 does not match the shape of the indexed tensor [853064] at index 0`

And this shape mismatch error can be reproduced in other dataset demos.

pengchongH, 2023-12-25:
Same problem. 

Thanks for your work. I also would like to know whether setting use_gaussian_splatting_densification=False means that we don't split the 3D gaussian.

Nik-V9, 2023-12-26:
Hi, Thanks for your interest in our work!

We don't use the official 3DGS Densification in SplaTAM. We don't currently support the `use_gaussian_splatting_densification` config. It was used in an old version of our code. We will try to fix this bug.

Ramseyous0109, 2023-12-27:
I get it. Thanks for your reply anyway!

pengchongH, 2023-12-28:
Thanks for your reply!

Buffyqsf, 2024-03-15:
Hello guys, I've meet a similar request. I haven't tried to run this densification strategy. But after reading the code, I don't know why this would happen. Have you fixed this bug?