spla-tam/SplaTAM #40: cyclonedds.core.DDSException

ansj11, 2023-12-20:
I am running the script bash bash_scripts/nerfcapture2dataset.bash configs/iphone/dataset.py，but got following error, what is the reason? By the way, my iPhone is iphone13, can I use it to run this script?
`sysctl: unknown oid 'net.core.rmem_max'
sysctl: unknown oid 'net.core.wmem_max'
1703056843.335096 [0]   42048241: config: //CycloneDDS/Domain/Internal/MinimumSocketReceiveBufferSize: setting moved to //CycloneDDS/Domain/Internal/SocketReceiveBufferSize[@min] (CYCLONEDDS_URI+0 line 1)
1703056843.335359 [0]   42048241: config: Domain/General/MulticastRecvNetworkInterfaceAddresses/#text: preferred {}
1703056843.335367 [0]   42048241: config: Domain/General/ExternalNetworkAddress/#text: auto {}
1703056843.335369 [0]   42048241: config: Domain/General/ExternalNetworkMask/#text: 0.0.0.0 {}
1703056843.335371 [0]   42048241: config: Domain/General/AllowMulticast/#text: default {}
1703056843.335372 [0]   42048241: config: Domain/General/MulticastTimeToLive/#text: 32 {}
1703056843.335374 [0]   42048241: config: Domain/General/DontRoute/#text: false {}
1703056843.335376 [0]   42048241: config: Domain/General/Transport/#text: udp {}
1703056843.335377 [0]   42048241: config: Domain/General/EnableMulticastLoopback/#text: true {}
1703056843.335379 [0]   42048241: config: Domain/General/MaxMessageSize/#text: 14720 B {}
1703056843.335380 [0]   42048241: config: Domain/General/MaxRexmitMessageSize/#text: 1456 B {}
1703056843.335381 [0]   42048241: config: Domain/General/FragmentSize/#text: 1344 B {}
1703056843.335383 [0]   42048241: config: Domain/General/RedundantNetworking/#text: false {}
1703056843.335384 [0]   42048241: config: Domain/General/EntityAutoNaming/#text: empty {}
1703056843.335386 [0]   42048241: config: Domain/General/EntityAutoNaming[@seed]: 981323711 2552571552 3736831470 568168365 1551115897 2539600070 2530733332 1219068320 {}
1703056843.335389 [0]   42048241: config: Domain/Sizing/ReceiveBufferSize/#text: 1 MiB {}
1703056843.335390 [0]   42048241: config: Domain/Sizing/ReceiveBufferChunkSize/#text: 128 KiB {}
1703056843.335392 [0]   42048241: config: Domain/Compatibility/StandardsConformance/#text: lax {}
1703056843.335393 [0]   42048241: config: Domain/Compatibility/ExplicitlyPublishQosSetToDefault/#text: false {}
1703056843.335395 [0]   42048241: config: Domain/Compatibility/ManySocketsMode/#text: single {}
1703056843.335396 [0]   42048241: config: Domain/Compatibility/AssumeRtiHasPmdEndpoints/#text: false {}
1703056843.335398 [0]   42048241: config: Domain/Discovery/Tag/#text:  {}
1703056843.335399 [0]   42048241: config: Domain/Discovery/ExternalDomainId/#text: 0 {}
1703056843.335400 [0]   42048241: config: Domain/Discovery/DSGracePeriod/#text: 30 s {}
1703056843.335402 [0]   42048241: config: Domain/Discovery/ParticipantIndex/#text: none {}
1703056843.335403 [0]   42048241: config: Domain/Discovery/MaxAutoParticipantIndex/#text: 9 {}
1703056843.335412 [0]   42048241: config: Domain/Discovery/SPDPMulticastAddress/#text: 239.255.0.1 {}
1703056843.335414 [0]   42048241: config: Domain/Discovery/SPDPInterval/#text: 30 s {}
1703056843.335415 [0]   42048241: config: Domain/Discovery/DefaultMulticastAddress/#text: auto {}
1703056843.335417 [0]   42048241: config: Domain/Discovery/Ports/Base/#text: 7400 {}
1703056843.335418 [0]   42048241: config: Domain/Discovery/Ports/DomainGain/#text: 250 {}
1703056843.335420 [0]   42048241: config: Domain/Discovery/Ports/ParticipantGain/#text: 2 {}
1703056843.335421 [0]   42048241: config: Domain/Discovery/Ports/MulticastMetaOffset/#text: 0 {}
1703056843.335423 [0]   42048241: config: Domain/Discovery/Ports/UnicastMetaOffset/#text: 10 {}
1703056843.335424 [0]   42048241: config: Domain/Discovery/Ports/MulticastDataOffset/#text: 1 {}
1703056843.335426 [0]   42048241: config: Domain/Discovery/Ports/UnicastDataOffset/#text: 11 {}
1703056843.335427 [0]   42048241: config: Domain/Discovery/EnableTopicDiscoveryEndpoints/#text: false {}
1703056843.335429 [0]   42048241: config: Domain/Discovery/LeaseDuration/#text: 10 s {}
1703056843.335432 [0]   42048241: config: Domain/Tracing/Category/#text: fatal,error,warning,info,config {1}
1703056843.335434 [0]   42048241: config: Domain/Tracing/OutputFile/#text: stdout {1}
1703056843.335435 [0]   42048241: config: Domain/Tracing/AppendToFile/#text: false {}
1703056843.335436 [0]   42048241: config: Domain/Tracing/PacketCaptureFile/#text:  {}
1703056843.335438 [0]   42048241: config: Domain/Internal/DeliveryQueueMaxSamples/#text: 256 {}
1703056843.335439 [0]   42048241: config: Domain/Internal/PrimaryReorderMaxSamples/#text: 128 {}
1703056843.335441 [0]   42048241: config: Domain/Internal/SecondaryReorderMaxSamples/#text: 128 {}
1703056843.335442 [0]   42048241: config: Domain/Internal/DefragUnreliableMaxSamples/#text: 4 {}
1703056843.335443 [0]   42048241: config: Domain/Internal/DefragReliableMaxSamples/#text: 16 {}
1703056843.335460 [0]   42048241: config: Domain/Internal/BuiltinEndpointSet/#text: writers {}
1703056843.335461 [0]   42048241: config: Domain/Internal/MeasureHbToAckLatency/#text: false {}
1703056843.335463 [0]   42048241: config: Domain/Internal/UnicastResponseToSPDPMessages/#text: true {}
1703056843.335464 [0]   42048241: config: Domain/Internal/SynchronousDeliveryPriorityThreshold/#text: 0 {}
1703056843.335466 [0]   42048241: config: Domain/Internal/SynchronousDeliveryLatencyBound/#text: inf {}
1703056843.335467 [0]   42048241: config: Domain/Internal/MaxParticipants/#text: 0 {}
1703056843.335468 [0]   42048241: config: Domain/Internal/AccelerateRexmitBlockSize/#text: 0 {}
1703056843.335470 [0]   42048241: config: Domain/Internal/RetransmitMerging/#text: never {}
1703056843.335471 [0]   42048241: config: Domain/Internal/RetransmitMergingPeriod/#text: 5 ms {}
1703056843.335473 [0]   42048241: config: Domain/Internal/HeartbeatInterval/#text: 100 ms {}
1703056843.335474 [0]   42048241: config: Domain/Internal/HeartbeatInterval[@min]: 5 ms {}
1703056843.335487 [0]   42048241: config: Domain/Internal/HeartbeatInterval[@minsched]: 20 ms {}
1703056843.335489 [0]   42048241: config: Domain/Internal/HeartbeatInterval[@max]: 8 s {}
1703056843.335490 [0]   42048241: config: Domain/Internal/MaxQueuedRexmitBytes/#text: 512 KiB {}
1703056843.335492 [0]   42048241: config: Domain/Internal/MaxQueuedRexmitMessages/#text: 200 {}
1703056843.335493 [0]   42048241: config: Domain/Internal/WriterLingerDuration/#text: 1 s {}
1703056843.335494 [0]   42048241: config: Domain/Internal/SocketReceiveBufferSize[@min]: 10 MiB {1}
1703056843.335496 [0]   42048241: config: Domain/Internal/SocketReceiveBufferSize[@max]: default {}
1703056843.335497 [0]   42048241: config: Domain/Internal/SocketSendBufferSize[@min]: 64 KiB {}
1703056843.335499 [0]   42048241: config: Domain/Internal/SocketSendBufferSize[@max]: default {}
[... 33 lines cut ...]
1703056843.335589 [0]   42048241: config: Domain/TCP/ReadTimeout/#text: 2 s {}
1703056843.335604 [0]   42048241: config: Domain/TCP/WriteTimeout/#text: 2 s {}
1703056843.335605 [0]   42048241: config: Domain/TCP/AlwaysUsePeeraddrForUnicast/#text: false {}
1703056843.335606 [0]   42048241: config: Domain[@Id]: 0 {1}
1703056843.335686 [0]   42048241: started at 1703056843.06335609 -- 2023-12-20 15:20:43+08:00
1703056843.335737 [0]   42048241: udp initialized
1703056843.335871 [0]   42048241: interfaces: lo0 udp/127.0.0.1(q1) en0 wireless udp/192.168.25.232(q9)
1703056843.335875 [0]   42048241: selected interfaces: en0 (index 12 priority 0)
1703056843.335876 [0]   42048241: presumed flaky multicast, use for SPDP only
1703056843.335960 [0]   42048241: ownip: udp/192.168.25.232
1703056843.335963 [0]   42048241: extmask: invalid/0 (not applicable)
1703056843.335964 [0]   42048241: SPDP MC: udp/239.255.0.1
1703056843.335965 [0]   42048241: default MC: udp/239.255.0.1
1703056843.335966 [0]   42048241: SSM support included
1703056843.336436 [0]   42048241: failed to increase socket receive buffer size to at least 10485760 bytes, current is 786896 bytes
1703056843.336436 [0]   42048241: failed to increase socket receive buffer size to at least 10485760 bytes, current is 786896 bytes
1703056843.336457 [0]   42048241: udp finalized
Traceback (most recent call last):
  File "scripts/nerfcapture2dataset.py", line 182, in <module>
    domain = Domain(domain_id=0, config=dds_config)
  File "/opt/miniconda3/lib/python3.8/site-packages/cyclonedds/domain.py", line 34, in __init__
    super().__init__(self._create_domain(dds_c_t.domainid(domain_id), config.encode("ascii")))
  File "/opt/miniconda3/lib/python3.8/site-packages/cyclonedds/core.py", line 181, in __init__
    raise DDSException(
cyclonedds.core.DDSException: [DDS_RETCODE_ERROR] Non specific error. Occurred upon initialisation of a cyclonedds.domain.Domain`

ansj11, 2023-12-20:
I have solved this problem in mac, but I run the script bash bash_scripts/nerfcapture2dataset.bash configs/iphone/dataset.py always "Waiting for frames...". I run cyclonedds ps print "Entities discovered: 2", but can not see the NeRFCapture APP. I am sure two device use same wifi, what can I do?
[image]
[image]

Nik-V9, 2023-12-26:
Hi, Thanks for trying out the code! Unfortunately, we aren't able to provide support with NeRFCapture connection issues.

I would recommend taking a look at the comments on this issue thread and adding any info you might deem relevant:
https://github.com/spla-tam/SplaTAM/issues/2

We will soon have an in-house variant that can stream videos through ROS2. That should provide a more seamless interface.

closed 2023-12-26 by Nik-V9

Shanzhaguoo, 2024-02-27:
> I have solved this problem in mac, but I run the script bash bash_scripts/nerfcapture2dataset.bash configs/iphone/dataset.py always "Waiting for frames...". I run cyclonedds ps print "Entities discovered: 2", but can not see the NeRFCapture APP. I am sure two device use same wifi, what can I do? ![截屏2023-12-20 16 25 08](https://private-user-images.githubusercontent.com/24786133/291828885-36113627-b4a5-4102-8976-aad9f8595e9c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwMDgyMjAsIm5iZiI6MTcwOTAwNzkyMCwicGF0aCI6Ii8yNDc4NjEzMy8yOTE4Mjg4ODUtMzYxMTM2MjctYjRhNS00MTAyLTg5NzYtYWFkOWY4NTk1ZTljLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI3VDA0MjUyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc1YjVhMTY3NWU1ZDRmNGVmMDI2YjI1Y2FhNjA1OWFhNDdhM2FkMzFhZjlhNzE0NjJiNjkxNmI1ZjZhYjQ1ZjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.NZgsKLj8QnMJhh_IIdXuv-FJBBPFPwoWKUCKIjV1-cA) ![截屏2023-12-20 16 25 20](https://private-user-images.githubusercontent.com/24786133/291828951-de328f42-d90e-497d-944d-b0b50858c08e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkwMDgyMjAsIm5iZiI6MTcwOTAwNzkyMCwicGF0aCI6Ii8yNDc4NjEzMy8yOTE4Mjg5NTEtZGUzMjhmNDItZDkwZS00OTdkLTk0NGQtYjBiNTA4NThjMDhlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI3VDA0MjUyMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI3OTYwYTUzODg4YWRlZWYyYmFhOTA4MDEwZDczMmNmZTFjNGUyMmE1NGRiZTc2ZGMwMDVhODY0YWY5MjRmODYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.F4f0xccJ0fW9sQJXVhYbUHNtxoGqESGlNt5oiQfkP6g)

Hi, I have encountered the same issue as mentioned above. Could I ask how you resolved it? By the way, I am running it on a Linux server.

sysctl: cannot stat /proc/sys/net/core/rmem_max: No such file or directory
sysctl: cannot stat /proc/sys/net/core/wmem_max: No such file or directory
1709007650.311879 [0]    python3: config: //CycloneDDS/Domain/Internal/MinimumSocketReceiveBufferSize: setting moved to //CycloneDDS/Domain/Internal/SocketReceiveBufferSize[@min] (CYCLONEDDS_URI+0 line 1)
1709007650.311975 [0]    python3: config: Domain/General/MulticastRecvNetworkInterfaceAddresses/#text: preferred {}
1709007650.311982 [0]    python3: config: Domain/General/ExternalNetworkAddress/#text: auto {}
1709007650.311988 [0]    python3: config: Domain/General/ExternalNetworkMask/#text: 0.0.0.0 {}
1709007650.311992 [0]    python3: config: Domain/General/AllowMulticast/#text: default {}
1709007650.311996 [0]    python3: config: Domain/General/MulticastTimeToLive/#text: 32 {}
1709007650.312000 [0]    python3: config: Domain/General/DontRoute/#text: false {}
1709007650.312005 [0]    python3: config: Domain/General/Transport/#text: udp {}
1709007650.312008 [0]    python3: config: Domain/General/EnableMulticastLoopback/#text: true {}
1709007650.312013 [0]    python3: config: Domain/General/MaxMessageSize/#text: 14720 B {}
1709007650.312017 [0]    python3: config: Domain/General/MaxRexmitMessageSize/#text: 1456 B {}
1709007650.312021 [0]    python3: config: Domain/General/FragmentSize/#text: 1344 B {}
1709007650.312025 [0]    python3: config: Domain/General/RedundantNetworking/#text: false {}
1709007650.312029 [0]    python3: config: Domain/General/EntityAutoNaming/#text: empty {}
1709007650.312034 [0]    python3: config: Domain/General/EntityAutoNaming[@seed]: 3859921027 2411168747 2224776029 368083721 2796016724 3974602444 172255704 1839214131 {}
1709007650.312041 [0]    python3: config: Domain/Sizing/ReceiveBufferSize/#text: 1 MiB {}
1709007650.312045 [0]    python3: config: Domain/Sizing/ReceiveBufferChunkSize/#text: 128 KiB {}
1709007650.312049 [0]    python3: config: Domain/Compatibility/StandardsConformance/#text: lax {}
1709007650.312053 [0]    python3: config: Domain/Compatibility/ExplicitlyPublishQosSetToDefault/#text: false {}
1709007650.312058 [0]    python3: config: Domain/Compatibility/ManySocketsMode/#text: single {}
1709007650.312062 [0]    python3: config: Domain/Compatibility/AssumeRtiHasPmdEndpoints/#text: false {}
1709007650.312067 [0]    python3: config: Domain/Discovery/Tag/#text:  {}
1709007650.312073 [0]    python3: config: Domain/Discovery/ExternalDomainId/#text: 0 {}
1709007650.312075 [0]    python3: config: Domain/Discovery/DSGracePeriod/#text: 30 s {}
1709007650.312080 [0]    python3: config: Domain/Discovery/ParticipantIndex/#text: none {}
1709007650.312084 [0]    python3: config: Domain/Discovery/MaxAutoParticipantIndex/#text: 9 {}
1709007650.312089 [0]    python3: config: Domain/Discovery/SPDPMulticastAddress/#text: 239.255.0.1 {}
1709007650.312093 [0]    python3: config: Domain/Discovery/SPDPInterval/#text: 30 s {}
1709007650.312099 [0]    python3: config: Domain/Discovery/DefaultMulticastAddress/#text: auto {}
1709007650.312104 [0]    python3: config: Domain/Discovery/Ports/Base/#text: 7400 {}
1709007650.312109 [0]    python3: config: Domain/Discovery/Ports/DomainGain/#text: 250 {}
1709007650.312112 [0]    python3: config: Domain/Discovery/Ports/ParticipantGain/#text: 2 {}
[... cut ...]
  File "scripts/nerfcapture2dataset.py", line 181, in <module>
    domain = Domain(domain_id=0, config=dds_config)
  File "/opt/conda/lib/python3.8/site-packages/cyclonedds/domain.py", line 34, in __init__
    super().__init__(self._create_domain(dds_c_t.domainid(domain_id), config.encode("ascii")))
  File "/opt/conda/lib/python3.8/site-packages/cyclonedds/core.py", line 181, in __init__
    raise DDSException(
cyclonedds.core.DDSException: [DDS_RETCODE_ERROR] Non specific error. Occurred upon initialisation of a cyclonedds.domain.Domain

Shanzhaguoo, 2024-02-27:
> I am running the script bash bash_scripts/nerfcapture2dataset.bash configs/iphone/dataset.py，but got following error, what is the reason? By the way, my iPhone is iphone13, can I use it to run this script? sysctl: unknown oid 'net.core.rmem_max' sysctl: unknown oid 'net.core.wmem_max' 1703056843.335096 [0] 42048241: config: //CycloneDDS/Domain/Internal/MinimumSocketReceiveBufferSize: setting moved to //CycloneDDS/Domain/Internal/SocketReceiveBufferSize[@min] (CYCLONEDDS_URI+0 line 1) 1703056843.335359 [0] 42048241: config: Domain/General/MulticastRecvNetworkInterfaceAddresses/#text: preferred {} 1703056843.335367 [0] 42048241: config: Domain/General/ExternalNetworkAddress/#text: auto {} 1703056843.335369 [0] 42048241: config: Domain/General/ExternalNetworkMask/#text: 0.0.0.0 {} 1703056843.335371 [0] 42048241: config: Domain/General/AllowMulticast/#text: default {} 1703056843.335372 [0] 42048241: config: Domain/General/MulticastTimeToLive/#text: 32 {} 1703056843.335374 [0] 42048241: config: Domain/General/DontRoute/#text: false {} 1703056843.335376 [0] 42048241: config: Domain/General/Transport/#text: udp {} 1703056843.335377 [0] 42048241: config: Domain/General/EnableMulticastLoopback/#text: true {} 1703056843.335379 [0] 42048241: config: Domain/General/MaxMessageSize/#text: 14720 B {} 1703056843.335380 [0] 42048241: config: Domain/General/MaxRexmitMessageSize/#text: 1456 B {} 1703056843.335381 [0] 42048241: config: Domain/General/FragmentSize/#text: 1344 B {} 1703056843.335383 [0] 42048241: config: Domain/General/RedundantNetworking/#text: false {} 1703056843.335384 [0] 42048241: config: Domain/General/EntityAutoNaming/#text: empty {} 1703056843.335386 [0] 42048241: config: Domain/General/EntityAutoNaming[@seed]: 981323711 2552571552 3736831470 568168365 1551115897 2539600070 2530733332 1219068320 {} 1703056843.335389 [0] 42048241: config: Domain/Sizing/ReceiveBufferSize/#text: 1 MiB {} 1703056843.335390 [0] 42048241: config: Domain/Sizing/ReceiveBufferChunkSize/#text: 128 KiB {} 1703056843.335392 [0] 42048241: config: Domain/Compatibility/StandardsConformance/#text: lax {} 1703056843.335393 [0] 42048241: config: Domain/Compatibility/ExplicitlyPublishQosSetToDefault/#text: false {} 1703056843.335395 [0] 42048241: config: Domain/Compatibility/ManySocketsMode/#text: single {} 1703056843.335396 [0] 42048241: config: Domain/Compatibility/AssumeRtiHasPmdEndpoints/#text: false {} 1703056843.335398 [0] 42048241: config: Domain/Discovery/Tag/#text: {} 1703056843.335399 [0] 42048241: config: Domain/Discovery/ExternalDomainId/#text: 0 {} 1703056843.335400 [0] 42048241: config: Domain/Discovery/DSGracePeriod/#text: 30 s {} 1703056843.335402 [0] 42048241: config: Domain/Discovery/ParticipantIndex/#text: none {} 1703056843.335403 [0] 42048241: config: Domain/Discovery/MaxAutoParticipantIndex/#text: 9 {} 1703056843.335412 [0] 42048241: config: Domain/Discovery/SPDPMulticastAddress/#text: 239.255.0.1 {} 1703056843.335414 [0] 42048241: config: Domain/Discovery/SPDPInterval/#text: 30 s {} 1703056843.335415 [0] 42048241: config: Domain/Discovery/DefaultMulticastAddress/#text: auto {} 1703056843.335417 [0] 42048241: config: Domain/Discovery/Ports/Base/#text: 7400 {} 1703056843.335418 [0] 42048241: config: Domain/Discovery/Ports/DomainGain/#text: 250 {} 1703056843.335420 [0] 42048241: config: Domain/Discovery/Ports/ParticipantGain/#text: 2 {} 1703056843.335421 [0] 42048241: config: Domain/Discovery/Ports/MulticastMetaOffset/#text: 0 {} 1703056843.335423 [0] 42048241: config: Domain/Discovery/Ports/UnicastMetaOffset/#text: 10 {} 1703056843.335424 [0] 42048241: config: Domain/Discovery/Ports/MulticastDataOffset/#text: 1 {} 1703056843.335426 [0] 42048241: config: Domain/Discovery/Ports/UnicastDataOffset/#text: 11 {} 1703056843.335427 [0] 42048241: config: Domain/Discovery/EnableTopicDiscoveryEndpoints/#text: false {} 1703056843.335429 [0] 42048241: config: Domain/Discovery/LeaseDuration/#text: 10 s {} 1703056843.335432 [0] 42048241: config: Domain/Tracing/Category/#text: fatal,error,warning,info,config {1} 1703056843.335434 [0] 42048241: config: Domain/Tracing/OutputFile/#text: stdout {1} 1703056843.335435 [0] 42048241: config: Domain/Tracing/AppendToFile/#text: false {} 1703056843.335436 [0] 42048241: config: Domain/Tracing/PacketCaptureFile/#text: {} 1703056843.335438 [0] 42048241: config: Domain/Internal/DeliveryQueueMaxSamples/#text: 256 {} 1703056843.335439 [0] 42048241: config: Domain/Internal/PrimaryReorderMaxSamples/#text: 128 {} 1703056843.335441 [0] 42048241: config: Domain/Internal/SecondaryReorderMaxSamples/#text: 128 {} 1703056843.335442 [0] 42048241: config: Domain/Internal/DefragUnreliableMaxSamples/#text: 4 {} 1703056843.335443 [0] 42048241: config: Domain/Internal/DefragReliableMaxSamples/#text: 16 {} 1703056843.335460 [0] 42048241: config: Domain/Internal/BuiltinEndpointSet/#text: writers {} 1703056843.335461 [0] 42048241: config: Domain/Internal/MeasureHbToAckLatency/#text: false {} 1703056843.335463 [0] 42048241: config: Domain/Internal/UnicastResponseToSPDPMessages/#text: true {} 1703056843.335464 [0] 42048241: config: Domain/Internal/SynchronousDeliveryPriorityThreshold/#text: 0 {} 1703056843.335466 [0] 42048241: config: Domain/Internal/SynchronousDeliveryLatencyBound/#text: inf {} 1703056843.335467 [0] 42048241: config: Domain/Internal/MaxParticipants/#text: 0 {} 1703056843.335468 [0] 42048241: config: Domain/Internal/AccelerateRexmitBlockSize/#text: 0 {} 1703056843.335470 [0] 42048241: config: [... cut ...]
File "scripts/nerfcapture2dataset.py", line 181, in
domain = Domain(domain_id=0, config=dds_config)
File "/opt/conda/lib/python3.8/site-packages/cyclonedds/domain.py", line 34, in init
super().init(self._create_domain(dds_c_t.domainid(domain_id), config.encode("ascii")))
File "/opt/conda/lib/python3.8/site-packages/cyclonedds/core.py", line 181, in init
raise DDSException(
cyclonedds.core.DDSException: [DDS_RETCODE_ERROR] Non specific error. Occurred upon initialisation of a cyclonedds.domain.Domain

ansj11, 2024-02-28:
you need increase the socket receive buffer size to at least 10485760 bytes, e.g. sysctl -w kern.ipc.maxsockbuf= 10485760 in mac. but I don't know the command in linux, you can see other issue or search in browser.

Zhaooyy, 2024-12-16:
> > I am running the script bash bash_scripts/nerfcapture2dataset.bash configs/iphone/dataset.py，but got following error, what is the reason? By the way, my iPhone is iphone13, can I use it to run this script? sysctl: unknown oid 'net.core.rmem_max' sysctl: unknown oid 'net.core.wmem_max' 1703056843.335096 [0] 42048241: config: //CycloneDDS/Domain/Internal/MinimumSocketReceiveBufferSize: setting moved to //CycloneDDS/Domain/Internal/SocketReceiveBufferSize[@min] (CYCLONEDDS_URI+0 line 1) 1703056843.335359 [0] 42048241: config: Domain/General/MulticastRecvNetworkInterfaceAddresses/#text: preferred {} 1703056843.335367 [0] 42048241: config: Domain/General/ExternalNetworkAddress/#text: auto {} 1703056843.335369 [0] 42048241: config: Domain/General/ExternalNetworkMask/#text: 0.0.0.0 {} 1703056843.335371 [0] 42048241: config: Domain/General/AllowMulticast/#text: default {} 1703056843.335372 [0] 42048241: config: Domain/General/MulticastTimeToLive/#text: 32 {} 1703056843.335374 [0] 42048241: config: Domain/General/DontRoute/#text: false {} 1703056843.335376 [0] 42048241: config: Domain/General/Transport/#text: udp {} 1703056843.335377 [0] 42048241: config: Domain/General/EnableMulticastLoopback/#text: true {} 1703056843.335379 [0] 42048241: config: Domain/General/MaxMessageSize/#text: 14720 B {} 1703056843.335380 [0] 42048241: config: Domain/General/MaxRexmitMessageSize/#text: 1456 B {} 1703056843.335381 [0] 42048241: config: Domain/General/FragmentSize/#text: 1344 B {} 1703056843.335383 [0] 42048241: config: Domain/General/RedundantNetworking/#text: false {} 1703056843.335384 [0] 42048241: config: Domain/General/EntityAutoNaming/#text: empty {} 1703056843.335386 [0] 42048241: config: Domain/General/EntityAutoNaming[@seed]: 981323711 2552571552 3736831470 568168365 1551115897 2539600070 2530733332 1219068320 {} 1703056843.335389 [0] 42048241: config: Domain/Sizing/ReceiveBufferSize/#text: 1 MiB {} 1703056843.335390 [0] 42048241: config: Domain/Sizing/ReceiveBufferChunkSize/#text: 128 KiB {} 1703056843.335392 [0] 42048241: config: Domain/Compatibility/StandardsConformance/#text: lax {} 1703056843.335393 [0] 42048241: config: Domain/Compatibility/ExplicitlyPublishQosSetToDefault/#text: false {} 1703056843.335395 [0] 42048241: config: Domain/Compatibility/ManySocketsMode/#text: single {} 1703056843.335396 [0] 42048241: config: Domain/Compatibility/AssumeRtiHasPmdEndpoints/#text: false {} 1703056843.335398 [0] 42048241: config: Domain/Discovery/Tag/#text: {} 1703056843.335399 [0] 42048241: config: Domain/Discovery/ExternalDomainId/#text: 0 {} 1703056843.335400 [0] 42048241: config: Domain/Discovery/DSGracePeriod/#text: 30 s {} 1703056843.335402 [0] 42048241: config: Domain/Discovery/ParticipantIndex/#text: none {} 1703056843.335403 [0] 42048241: config: Domain/Discovery/MaxAutoParticipantIndex/#text: 9 {} 1703056843.335412 [0] 42048241: config: Domain/Discovery/SPDPMulticastAddress/#text: 239.255.0.1 {} 1703056843.335414 [0] 42048241: config: Domain/Discovery/SPDPInterval/#text: 30 s {} 1703056843.335415 [0] 42048241: config: Domain/Discovery/DefaultMulticastAddress/#text: auto {} 1703056843.335417 [0] 42048241: config: Domain/Discovery/Ports/Base/#text: 7400 {} 1703056843.335418 [0] 42048241: config: Domain/Discovery/Ports/DomainGain/#text: 250 {} 1703056843.335420 [0] 42048241: config: Domain/Discovery/Ports/ParticipantGain/#text: 2 {} 1703056843.335421 [0] 42048241: config: Domain/Discovery/Ports/MulticastMetaOffset/#text: 0 {} 1703056843.335423 [0] 42048241: config: Domain/Discovery/Ports/UnicastMetaOffset/#text: 10 {} 1703056843.335424 [0] 42048241: config: Domain/Discovery/Ports/MulticastDataOffset/#text: 1 {} 1703056843.335426 [0] 42048241: config: Domain/Discovery/Ports/UnicastDataOffset/#text: 11 {} 1703056843.335427 [0] 42048241: config: Domain/Discovery/EnableTopicDiscoveryEndpoints/#text: false {} 1703056843.335429 [0] 42048241: config: Domain/Discovery/LeaseDuration/#text: 10 s {} 1703056843.335432 [0] 42048241: config: Domain/Tracing/Category/#text: fatal,error,warning,info,config {1} 1703056843.335434 [0] 42048241: config: Domain/Tracing/OutputFile/#text: stdout {1} 1703056843.335435 [0] 42048241: config: Domain/Tracing/AppendToFile/#text: false {} 1703056843.335436 [0] 42048241: config: Domain/Tracing/PacketCaptureFile/#text: {} 1703056843.335438 [0] 42048241: config: Domain/Internal/DeliveryQueueMaxSamples/#text: 256 {} 1703056843.335439 [0] 42048241: config: Domain/Internal/PrimaryReorderMaxSamples/#text: 128 {} 1703056843.335441 [0] 42048241: config: Domain/Internal/SecondaryReorderMaxSamples/#text: 128 {} 1703056843.335442 [0] 42048241: config: Domain/Internal/DefragUnreliableMaxSamples/#text: 4 {} 1703056843.335443 [0] 42048241: config: Domain/Internal/DefragReliableMaxSamples/#text: 16 {} 1703056843.335460 [0] 42048241: config: Domain/Internal/BuiltinEndpointSet/#text: writers {} 1703056843.335461 [0] 42048241: config: Domain/Internal/MeasureHbToAckLatency/#text: false {} 1703056843.335463 [0] 42048241: config: Domain/Internal/UnicastResponseToSPDPMessages/#text: true {} 1703056843.335464 [0] 42048241: config: Domain/Internal/SynchronousDeliveryPriorityThreshold/#text: 0 {} 1703056843.335466 [0] 42048241: config: Domain/Internal/SynchronousDeliveryLatencyBound/#text: inf {} 1703056843.335467 [0] 42048241: config: Domain/Internal/MaxParticipants/#text: 0 {} 1703056843.335468 [0] 42048241: config: Domain/Internal/AccelerateRexmitBlockSize/#text: 0 {} 1703056843.335470 [0] 42048241: confi [... cut ...] code（“asphalt”） File "/opt/conda/lib/python3.8/site-packages/cyclonedds/core.py", line 181, in init File“/opt/conda/lib/python3.8/site-packages/cyclonedds/core.py“，line 181，in init raise DDSException(  引发DDSException（ cyclonedds.core.DDSException: [DDS_RETCODE_ERROR] Non specific error. Occurred upon initialisation of a cyclonedds.domain.Domain cyclonedds.core.DDSException：[DDS_RETCODE_ERROR]非特定错误。在初始化cyclonedds.domain.Domain时出现

请问解决了吗，我也在Linux系统遇到了相关问题，尝试设置/proc/sys/net/core/rmem_max又显示没有相关文件。