spla-tam/SplaTAM #44: Depth image obtained using Stereo images

darshanAtSwaayatt, 2023-12-27:
Can I use this code to generate 3D reconstruction using Depth images obtained using Stereo images?

Nik-V9, 2023-12-28:
Hi, yes, it is possible to extend our current RGB-D implementation to work with a Stereo setup. Support for two images as input will require significant restructuring of the code. Hence, this is currently not on our radar.

darshanAtSwaayatt, 2023-12-28:
Hi @Nik-V9 ,

I have a stereo data and used "https://wiki.ros.org/stereo_image_proc" ROS package to generate depth images. 
I gave this data as input to the Splatam code.
I updated the code with necessary scripts like the data.yaml, config script, dataset class, etc. but the output received is not proper.
For the below scene (data is captured using the stereo setup mounted on a car. So the camera motion is in the forward direction only)
[image]

Corresponding depth image
[image]

the output generated was
FRONT VIEW
[image]

SIDE VIEW
[image]

TOP VIEW
[image]

But using Iphone 12 pro data, the output was proper.

The config files updated by me for above data are in below zip:
[Splatam_updated_files.zip](https://github.com/spla-tam/SplaTAM/files/13783386/Splatam_updated_files.zip)

Also wanted to understand how I can find "png_depth_scale" parameter for my data?
As far as my understanding I can use below formula:
[image]

where 
- f is the focal length
- B is the baseline distance between the stereo cameras,
- D_image :  is the pixel value in the depth image.
- Z_meters : Real world depth in meters.

Z_meters will be the png_depth_scale for my data. But this will not be constant, as Z_meters is inversely proportional to D_image.

Can you tell what I am doing wrong?
Any suggestions to fix my problem.

Thanks and Regards,
@darshanAtSwaayatt

darshanAtSwaayatt, 2024-01-11:
> Hi @Nik-V9 ,
> 
> I have a stereo data and used "https://wiki.ros.org/stereo_image_proc" ROS package to generate depth images. I gave this data as input to the Splatam code. I updated the code with necessary scripts like the data.yaml, config script, dataset class, etc. but the output received is not proper. For the below scene (data is captured using the stereo setup mounted on a car. So the camera motion is in the forward direction only) ![left_4881](https://private-user-images.githubusercontent.com/130635154/293137020-6d8bc584-4497-4ebe-bf34-f6607d980855.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5NzE0MDEsIm5iZiI6MTcwNDk3MTEwMSwicGF0aCI6Ii8xMzA2MzUxNTQvMjkzMTM3MDIwLTZkOGJjNTg0LTQ0OTctNGViZS1iZjM0LWY2NjA3ZDk4MDg1NS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMVQxMTA1MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NWZiN2UxNDUwYjRmNmNlODRlYzZjYzg3ZWQwMmUyODVjNjRhZDI5ZTRjMjFkY2ZjNDliYjhjYWYxMDdhZDhjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.cX9hqm45sZnzIVYfytNgrn8PTS3WjKIgv--W2paWHEQ)
> 
> Corresponding depth image ![disp_4881](https://private-user-images.githubusercontent.com/130635154/293137246-f4f97eb2-dea4-4818-885f-7bf4700f2d1c.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5NzE0MDEsIm5iZiI6MTcwNDk3MTEwMSwicGF0aCI6Ii8xMzA2MzUxNTQvMjkzMTM3MjQ2LWY0Zjk3ZWIyLWRlYTQtNDgxOC04ODVmLTdiZjQ3MDBmMmQxYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMVQxMTA1MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1lYjcwZjQ0NjYyODYxM2JjOWZjMTg4ZjE1OTA5YWVmMTliNGE4NzY1MzY5ZWM4ZWZmODAzMjhjMTc2ODY2MDQ5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.NOGi06ZPJ5EJTH_KLO0Q1yMW8DsoXir7w7GPZwivLa4)
> 
> the output generated was FRONT VIEW ![image](https://private-user-images.githubusercontent.com/130635154/293138627-5b9e3323-2648-462e-92ef-dafbe6e45297.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5NzE0MDEsIm5iZiI6MTcwNDk3MTEwMSwicGF0aCI6Ii8xMzA2MzUxNTQvMjkzMTM4NjI3LTViOWUzMzIzLTI2NDgtNDYyZS05MmVmLWRhZmJlNmU0NTI5Ny5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMVQxMTA1MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jOTUyODQwYzA0NDJkYzBkMmQwMTBhMDFhMjY1NGYyODhkOGJjOWVlY2QxZjBiOGE5NzEwNTEwOWI1OGViNzgyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.5mcZxMw8GuUwv6BE4XxeamIq5TS_EP_o9zYhVuZk9Lk)
> 
> SIDE VIEW ![image](https://private-user-images.githubusercontent.com/130635154/293138768-d0df32b0-d2ab-4942-8a4f-1e8cacd0d043.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5NzE0MDEsIm5iZiI6MTcwNDk3MTEwMSwicGF0aCI6Ii8xMzA2MzUxNTQvMjkzMTM4NzY4LWQwZGYzMmIwLWQyYWItNDk0Mi04YTRmLTFlOGNhY2QwZDA0My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMVQxMTA1MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1iNDE5MTQ4ZmM1NjI5MTRlMzdjMjViNDIyZDJiNDIyZmQ1YTlkYWQwNGQ2NzFhZGMyMGE4MGJjMDNiZjY4ZmNiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Zli5YChqcdomFfn-RH-ZBWrq_2Hv9fTm8qnxcCfxP-M)
> 
> TOP VIEW ![image](https://private-user-images.githubusercontent.com/130635154/293138840-67553dd5-b48b-4ed6-8c37-2b48c3036f3d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5NzE0MDEsIm5iZiI6MTcwNDk3MTEwMSwicGF0aCI6Ii8xMzA2MzUxNTQvMjkzMTM4ODQwLTY3NTUzZGQ1LWI0OGItNGVkNi04YzM3LTJiNDhjMzAzNmYzZC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMTExJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDExMVQxMTA1MDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jODIzZGFkYjg0NzQxMmEwODZmZDcwOTA5NTI4N2IzOTY4YzliZjFmNWRkMjJiMzU3ZWJjOWY2NWNiZWIwNDNlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.iBTfwv_jWD0NwgGiCL2Eka3B31LC_8W6_ouD27G2pnU)
> 
> But using Iphone 12 pro data, the output was proper.
> 
> The config files updated by me for above data are in below zip: [Splatam_updated_files.zip](https://github.com/spla-tam/SplaTAM/files/13783386/Splatam_updated_files.zip)
> 
> Also wanted to understand how I can find "png_depth_scale" parameter for my data? As far as my understanding I can use below formula: ![image](https://private-user-images.githubusercontent.com/130635154/293140359-d63fb579-9c14-46fe-94d4-e2dc037e2699.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQ5NzE0MDEsIm5iZiI6MTcwNDk3MTEwMSwicGF0aCI6Ii8xMzA2MzUxNTQvMjkzMTQwMzU5LWQ2M2ZiNTc5LTljMTQtNDZmZS05NGQ0LWUyZGMwMzdlMjY5OS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtS [... cut ...] hwY7ig5Ssoz8)
> 
> where
> 
> * f is the focal length
> * B is the baseline distance between the stereo cameras,
> * D_image :  is the pixel value in the depth image.
> * Z_meters : Real world depth in meters.
> 
> Z_meters will be the png_depth_scale for my data. But this will not be constant, as Z_meters is inversely proportional to D_image.
> 
> Can you tell what I am doing wrong? Any suggestions to fix my problem.
> 
> Thanks and Regards, @darshanAtSwaayatt

Hi @Nik-V9 , Can you please help.

Nik-V9, 2024-01-23:
Hi, if the depth image you are saving has the pixel values in meters, the `png_depth_scale` will be 1.

Essentially `png_depth_scale` is a parameter to convert pixel values of a depth image to metric depth. In your case, after computing the metric depth if you are saving the image as is, the `png_depth_scale` will be 1.

Also based on the stereo depth image shared by you, there seems to be a lot of holes in the depth map. Without good multi-view coverage, I'm unsure how well the current version of SplaTAM will perform on the data.