spla-tam/SplaTAM #45: Render RGB and depth in one shot

Willyzw, 2023-12-29:
Hi, I noticed the current code renders RGB and depth separately using different `colors_precomp`, I am curious whether it is possible to render them in just one shot. What modifications need to be made? e.g. on the forward and backward pass

ljjTYJR, 2023-12-31:
Modify the code in [diff-gaussian-rasterization-w-depth](https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth),
just view the depth as one channel of the colors

Willyzw, 2024-01-01:
> Modify the code in [diff-gaussian-rasterization-w-depth](https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth), just view the depth as one channel of the colors

indeed this would be a simple but effective way! Thanks for your suggestion 👍

Willyzw, 2024-01-01:
closed as question got replied

closed 2024-01-01 by Willyzw