spla-tam/SplaTAM #47: wandb error

Runhane, 2024-01-02:
When I was running "python scripts/splatam.py configs/tum/splatam.py",  I encountered the following error.
Loaded Config:
{'workdir': './experiments/TUM', 'run_name': 'freiburg1_desk_seed0', 'seed': 0, 'primary_device': 'cuda:0', 'map_every': 1, 'keyframe_every': 5, 'mapping_window_size': 20, 'report_global_progress_every': 500, 'eval_every': 500, 'scene_radius_depth_ratio': 2, 'mean_sq_dist_method': 'projective', 'report_iter_progress': False, 'load_checkpoint': False, 'checkpoint_time_idx': 0, 'save_checkpoints': False, 'checkpoint_interval': 100, 'use_wandb': True, 'wandb': {'entity': 'theairlab', 'project': 'SplaTAM', 'group': 'TUM', 'name': 'freiburg1_desk_seed0', 'save_qual': False, 'eval_save_qual': True}, 'data': {'basedir': './data/TUM_RGBD', 'gradslam_data_cfg': './configs/data/TUM/freiburg1_desk.yaml', 'sequence': 'rgbd_dataset_freiburg1_desk', 'desired_image_height': 480, 'desired_image_width': 640, 'start': 0, 'end': -1, 'stride': 1, 'num_frames': -1}, 'tracking': {'use_gt_poses': False, 'forward_prop': True, 'num_iters': 200, 'use_sil_for_loss': True, 'sil_thres': 0.99, 'use_l1': True, 'ignore_outlier_depth_loss': False, 'use_uncertainty_for_loss_mask': False, 'use_uncertainty_for_loss': False, 'use_chamfer': False, 'loss_weights': {'im': 0.5, 'depth': 1.0}, 'lrs': {'means3D': 0.0, 'rgb_colors': 0.0, 'unnorm_rotations': 0.0, 'logit_opacities': 0.0, 'log_scales': 0.0, 'cam_unnorm_rots': 0.002, 'cam_trans': 0.002}, 'use_depth_loss_thres': False, 'depth_loss_thres': 100000, 'visualize_tracking_loss': False}, 'mapping': {'num_iters': 30, 'add_new_gaussians': True, 'sil_thres': 0.5, 'use_l1': True, 'use_sil_for_loss': False, 'ignore_outlier_depth_loss': False, 'use_uncertainty_for_loss_mask': False, 'use_uncertainty_for_loss': False, 'use_chamfer': False, 'loss_weights': {'im': 0.5, 'depth': 1.0}, 'lrs': {'means3D': 0.0001, 'rgb_colors': 0.0025, 'unnorm_rotations': 0.001, 'logit_opacities': 0.05, 'log_scales': 0.001, 'cam_unnorm_rots': 0.0, 'cam_trans': 0.0}, 'prune_gaussians': True, 'pruning_dict': {'start_after': 0, 'remove_big_after': 0, 'stop_after': 20, 'prune_every': 20, 'removal_opacity_threshold': 0.005, 'final_removal_opacity_threshold': 0.005, 'reset_opacities': False, 'reset_opacities_every': 500}, 'use_gaussian_splatting_densification': False, 'densify_dict': {'start_after': 500, 'remove_big_after': 3000, 'stop_after': 5000, 'densify_every': 100, 'grad_thresh': 0.0002, 'num_to_split_into': 2, 'removal_opacity_threshold': 0.005, 'final_removal_opacity_threshold': 0.005, 'reset_opacities_every': 3000}}, 'viz': {'render_mode': 'color', 'offset_first_viz_cam': True, 'show_sil': False, 'visualize_cams': True, 'viz_w': 600, 'viz_h': 340, 'viz_near': 0.01, 'viz_far': 100.0, 'view_scale': 2, 'viz_fps': 5, 'enter_interactive_post_online': False}}
wandb: (1) Create a W&B account
wandb: (2) Use an existing W&B account
wandb: (3) Don't visualize my results
wandb: Enter your choice: 2
wandb: You chose 'Use an existing W&B account'
wandb: Logging into wandb.ai. (Learn how to deploy a W&B server locally: https://wandb.me/wandb-server)
wandb: You can find your API key in your browser here: https://wandb.ai/authorize
wandb: Paste an API key from your profile and hit enter, or press ctrl+c to quit:
wandb: Appending key for api.wandb.ai to your netrc file: /home/zrh/.netrc
wandb: ERROR Error while calling W&B API: project not found (<Response [404]>)
Problem at: /home/zrh/project/SplaTAM/scripts/splatam.py 472 rgbd_slam
wandb: ERROR It appears that you do not have permission to access the requested resource. Please reach out to the project owner to grant you access. If you have the correct permissions, verify that there are no issues with your networking setup.(Error 404: Not Found)
Traceback (most recent call last):
  File "/home/zrh/project/SplaTAM/scripts/splatam.py", line 1007, in <module>
    rgbd_slam(experiment.config)
  File "/home/zrh/project/SplaTAM/scripts/splatam.py", line 472, in rgbd_slam
    wandb_run = wandb.init(project=config['wandb']['project'],
  File "/home/zrh/source/anaconda3/envs/test/lib/python3.10/site-packages/wandb/sdk/wandb_init.py", line 1189, in init
    raise e
  File "/home/zrh/source/anaconda3/envs/test/lib/python3.10/site-packages/wandb/sdk/wandb_init.py", line 1170, in init
    run = wi.init()
  File "/home/zrh/source/anaconda3/envs/test/lib/python3.10/site-packages/wandb/sdk/wandb_init.py", line 781, in init
    raise error
wandb.errors.CommError: It appears that you do not have permission to access the requested resource. Please reach out to the project owner to grant you access. If you have the correct permissions, verify that there are no issues with your networking setup.(Error 404: Not Found)

closed 2024-01-03 by Runhane

jinglun531, 2024-01-19:
@Runhane Hey did you resolve this issue? If so could you please share the way you solved it?

Runhane, 2024-01-19:
这是来自QQ邮箱的假期自动回复邮件。你好，我已收到你的邮件 ：）

wangjinhoon, 2024-01-22:
> @Runhane Hey did you resolve this issue? If so could you please share the way you solved it?

delete "entity=config['wandb']['entity'],"