spla-tam/SplaTAM #52: Confusion about 'transformed_params2depthplussilhouette' function

RPFey, 2024-01-10:
Thank you for sharing the code. I am confused about the `transformed_params2depthplussilhouette` function here.

```python
def transformed_params2depthplussilhouette(params, w2c, transformed_pts):
    rendervar = {
        'means3D': transformed_pts,
        'colors_precomp': get_depth_and_silhouette(transformed_pts, w2c),
        'rotations': F.normalize(params['unnorm_rotations']),
        'opacities': torch.sigmoid(params['logit_opacities']),
        'scales': torch.exp(torch.tile(params['log_scales'], (1, 3))),
        'means2D': torch.zeros_like(params['means3D'], requires_grad=True, device="cuda") + 0
    }
    return rendervar
```

Since in the `transform_to_frame` function, `transformed_pts` are the points transformed into current camera frame. Why do we need to transform it again in the code ` get_depth_and_silhouette(transformed_pts, w2c)` ? Shouldn't we just use the `transformed_pts` to compute the depth ?

Buffyqsf, 2024-01-10:
I think it's related to the render camera setting. The gaussian render `Renderer(raster_settings = curr_data['cam']`, so the render camera's view is always from the first frame's position (call it as view camera). Usually it's identity, which is easy to understand. But when it's not identity, the depth value from scene to view camera used be adjusted. 
I may not express it very well, but I think the key is there is a view camera.

Nik-V9, 2024-01-10:
Hi @RPFey, Thanks for your question! @Buffyqsf is right.

The input `w2c` for the `transformed_params2depthplussilhouette` function is the `first_frame_w2c`, the viewing camera. Ideally, if all the poses are with respect to the first frame (as is the case with SplaTAM), the input `w2c` would be identity. However, to make sure the function is general and works with any viewing camera definition, we convert the world frame Gaussians to the viewing camera frame to compute the depth:
https://github.com/spla-tam/SplaTAM/blob/a0bda58dd6fbf3e2ad31e40adc48514923bec4c0/utils/slam_helpers.py#L172

RPFey, 2024-01-15:
Thank you for your replies. It answers my questions !

closed 2024-01-16 by Nik-V9

whwh747, 2024-12-16:
@Buffyqsf hello，I am still confused about why the view matrix and projection matrix of the rendering camera remain the same in every frame. What is the purpose of these two matrices?

Buffyqsf, 2024-12-16:
@whwh747 I'm sorry that I don't follow this for a long time. I remember it because SplaTAM transformed the scene instead of the camera.

medsadw, 2025-01-24:
> I think it's related to the render camera setting. The gaussian render `Renderer(raster_settings = curr_data['cam']`, so the render camera's view is always from the first frame's position (call it [...]

我有个问题想要咨询，在'colors_precomp': get_depth_and_silhouette(params['means3D'], w2c)中的 params['means3D']是之前经过函数transform_to_frame(params, time_idx, gaussians_grad=False, camera_grad=False)处理的，即这里params['means3D']目前是在相机坐标系下的三维点（命名为Pc），我看到w2c的定义为第一帧的相机的姿态（T{c_1}{w}）,但是这个

medsadw, 2025-01-24:
> I think it's related to the render camera setting. The gaussian render `Renderer(raster_settings = curr_data['cam']`, so the render camera's view is always from the first frame's position (call it [...]

I am sorry to disturb you during your busy schedule, but I still have questions that need consulting. In the line `'colors_precomp': get_depth_and_silhouette(params['means3D'], w2c)` ,  `params['means3D']` has been processed by the `transform_to_frame(params, time_idx, gaussians_grad=False, camera_grad=False)`.  As I noticed, w2c is defined as the pose of the camera in the first frame（T{c_1}{w}）, In the code snippet for `get_depth_and_silhouette(params['means3D'], w2c)`  
```
def get_depth_and_silhouette(pts_3D, w2c):
    pts4 = torch.cat((pts_3D, torch.ones_like(pts_3D[:, :1])), dim=-1)
    pts_in_cam = (w2c @ pts4.transpose(0, 1)).transpose(0, 1)
```
The operation `w2c @ pts4.transpose(0, 1)`  seems to multiply the pose of the first frame T{c_1}{w} onto the 3D points of Gaussian centers. 
I believe that if original `params['means3D']` refers to the Gaussian 3D points P{w} in the world coordinate system, it should first be transformed by the pose of the first frame's camera w2c (i.e., multiplied by T{c1}{w}, converting it into the first frame's coordinate system), and then undergo `transform_to_frame(...)` (i.e., multiplied by T{c}{c1}), so that the 3D points are converted into the camera coordinate system, making the retrieved depth truly represent the depth in the camera's coordinate system. It shouldn't be the case where we perform `transform_to_frame(...)` first, followed by matrix multiplication with the pose of the first frame's camera w2c, because matrix multiplication does not satisfy the commutative law of multiplication.         
 (i.e., T{c|c1}T{c1|w} Pw!=T{c1|w}T{c|c1}Pw).

FDF521X, 2025-03-04:
> I am sorry to disturb you during your busy schedule, but I still have questions that need consulting. In the line `'colors_precomp': get_depth_and_silhouette(params['means3D'], w2c)` , [...]

I have the same confusion.

BW-Zhang05, 2025-07-08:
> I am sorry to disturb you during your busy schedule, but I still have questions that need consulting. In the line `'colors_precomp': get_depth_and_silhouette(params['means3D'], w2c)` , [...]

The pose in `transform_to_frame(params, time_idx, gaussians_grad=False, camera_grad=False)` is the current frame's pose and it stands w2c：world to camera