spla-tam/SplaTAM #54: Regarding depth backpropagation

eriksandstroem, 2024-01-11:
Hi,
First of all, thanks a lot for providing this repository open source!

My question should be fairly easy to answer: In the paper (and code) you use depth supervision during mapping and tracking and I have confirmed that this works as intended, however, when looking at the submodule library [https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/tree/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110](url) it explicitly says "Note that the backward pass for the depth has not been implemented, so it won't work for training with depth ground-truth". This makes me confused as to why the code still seems to implement the depth supervision.

Would be very grateful for insights on this!

Cheers,
Erik

Nik-V9, 2024-01-11:
Hi, Thanks for your interest in our work!

Instead of using CUDA for the depth backpropagation, we use PyTorch to render depth as a channel of dummy color in the following way described here: https://github.com/spla-tam/SplaTAM/issues/26#issuecomment-1855175165.

This enables us to get backprop gradients from PyTorch. However, the drawback is that we have to render twice, as mentioned in this issue: https://github.com/spla-tam/SplaTAM/issues/45

Please also see: https://x.com/janusch_patas/status/1735436291624317427?s=20

Hence, the speed of our system can be further enhanced by rendering depth through the CUDA code simultaneously.

eriksandstroem, 2024-01-11:
Thanks for the quick reply!

closed 2024-01-11 by eriksandstroem