spla-tam/SplaTAM #55: recon error

zhuhu00, 2024-01-15:
This is a very good job, and when I was experimenting with the replica dataset, I encountered the following error in the Reconstruction section, but I couldn't find a solution. The open3d==0.16.0 I use. How can I fix it?
```bash
Traceback (most recent call last):
  File "/root/code/viz_scripts/online_recon.py", line 361, in <module>
    visualize(scene_path, viz_cfg)
  File "/root/code/viz_scripts/online_recon.py", line 223, in visualize
    view_control.convert_from_pinhole_camera_parameters(cparams, allow_arbitrary=True)
AttributeError: 'NoneType' object has no attribute 'convert_from_pinhole_camera_parameters'
```

xiaoyaocoding, 2024-01-16:
I encountered the same issue as you, and it was because I didn't enable remote desktop while running code on the server. After enabling the desktop, the problem was automatically resolved. I hope this helps you.

xiaoyaocoding, 2024-01-16:
[image]
the difference is I run “python viz_scripts/final_recon.py configs/replica/splatam.py”

zhuhu00, 2024-01-16:
> ![image](https://private-user-images.githubusercontent.com/102663190/296953022-50da7166-45c4-447e-82f2-be1ce08c9747.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Thanks a lot.

closed 2024-01-16 by zhuhu00