spla-tam/SplaTAM #56: Novel view

zero-joke, 2024-01-15:
I am very excited about your work .  How can I render a picture in novel view?

Nik-V9, 2024-01-22:
Hi, Thanks for your interest in our work!

The following function is what we use for Replica Novel View Synthesis: https://github.com/spla-tam/SplaTAM/blob/a0bda58dd6fbf3e2ad31e40adc48514923bec4c0/utils/eval_helpers.py#L623

Similar code can be used to render novel views from a SplaTAM reconstruction. The main steps are as follows:

- Initialize pose for the novel view (this is with respect to the first frame used for SLAM; this is the origin of the reconstruction).
- Render the 3D Gaussian Map from the novel camera pose.

Our interactive visualizer also does the same thing: https://github.com/spla-tam/SplaTAM/blob/a0bda58dd6fbf3e2ad31e40adc48514923bec4c0/viz_scripts/final_recon.py#L241

Nik-V9, 2024-02-03:
Closing Issue due to inactivity.

closed 2024-02-03 by Nik-V9