spla-tam/SplaTAM #57: How can i get a reconstruction mesh

anghaocheng, 2024-01-16:
Thank you very much for sharing your excellent work. I got very good rendering results after running the replica dataset, but I did not find the reconstructed 3D mesh. Is there any way  I can get a reconstruction mesh?

JayKarhade, 2024-01-21:
Hi @anghaocheng , thanks for your interest. 

We current do not support 3D mesh reconstructions from gaussians yet, however you can take a look at some recent work : https://imagine.enpc.fr/~guedona/sugar/

Also i would encourage to take a look at this issue for more details: https://github.com/spla-tam/SplaTAM/issues/33#issuecomment-1869540660

Nik-V9, 2024-01-21:
Closing this issue since it's similar to #33. Please post updates there.

closed 2024-01-21 by Nik-V9