spla-tam/SplaTAM #58: How to use 3D gaussian-based  map for navigation?

jeezrick, 2024-01-17:
Thank you very much for sharing your excellent work. 
 
I'm curious about the practical applications of using the trained Gaussians as a map for navigation. Can they be effectively employed in navigation? 
Or are they primarily useful as a collection of 2D images, distinct from representing a true 3D scene as a point cloud?

I appreciate your insights into the potential navigational applications of trained Gaussians and how they can be utilized in this context.

Nik-V9, 2024-01-22:
Hi, Thanks for your interest in 3D Gaussians!

Yes, 3D Gaussians can be employed in navigation and obstacle avoidance. The specifics depend on the goals and design of the pipeline. At the end of the day, 3D Gaussians are a way to represent a scene for a robot. Depending on its characteristics, this representation can be used in many for navigation.

Off the top of my head, here are three examples of how 3D Gaussians might help in navigation:

- One nice property of 3D Gaussian Splatting is that the alpha-composite (accumulated opacity for each pixel, silhouette) directly indicates how well Gaussians along a particular ray are optimized. Hence, this is a form of aleatoric uncertainty. This can be used to guide path planning and coverage.
- If you are building a 3D Gaussian map with embeddings on the fly, such a scene representation can be queried for navigation tasks. For example, this would be similar to ConceptFusion (https://concept-fusion.github.io/) and ConceptGraphs (https://concept-graphs.github.io/).
- If you have a stereo setup or sparse depth, a 3D Gaussian-based representation can help you densify a reconstruction and avoid obstacles.

I also see that you've opened an issue here: https://github.com/minghanqin/LangSplat/issues/6. Since this is related to my second example, I'll answer this:

Let's say you want to query LangSplat for 'TV'. Then, you essentially will use a model to encode the text 'TV' to an embedding. Then, you will use this embedding and match it with the embeddings of the 3D Gaussian Map. Places on the map with high similarity indicate potential locations for a TV. In this way, you can query a 3D scene representation. Please refer to ConceptFusion for more details.

jeezrick, 2024-01-26:
Thanks a lot for the reply, I will read the papers you mentioned. 

About my question with LanSplat, this is my understanding: As the paper of LangSplat says, they use autoencoder to compress dim 512 clip embedding to dim 3 latent. So, the gaussians itself only attach dim 3 latent, right? I don't know if the compressed dim 3 latent can still query each other. But if they can't, you just have to render the 3d gaussians to a 2d image with dim 512 clip embeddings attached, then you can query it. But the problem is, within all the camera poses, we don't which one makes the 'TV' visiable. 

I have asked this question several time at some places. So I guess my question is, can we directly quering the 3d gaussians without rendering it as a image. Cause if we can't, I can see how LangSplat can be used for localization(but in the paper, they showed result about localization).

jeezrick, 2024-01-26:
I have been studying the Splatam v1 code from you guys. 

And I have some question about it, I wonder if you can shed some light. 

In calculating silhouette

```python

def get_depth_and_silhouette(pts_3D, w2c):
    """
    Function to compute depth and silhouette for each gaussian.
    These are evaluated at gaussian center.
    """
    # Depth of each gaussian center in camera frame
    pts4 = torch.cat((pts_3D, torch.ones_like(pts_3D[:, :1])), dim=-1) # TODO: slow?
    pts_in_cam = (w2c @ pts4.transpose(0, 1)).transpose(0, 1)
    depth_z = pts_in_cam[:, 2].unsqueeze(-1) # [num_gaussians, 1]
    depth_z_sq = torch.square(depth_z) # [num_gaussians, 1]

    # Depth and Silhouette
    depth_silhouette = torch.zeros((pts_3D.shape[0], 3)).cuda().float()
    depth_silhouette[:, 0] = depth_z.squeeze(-1)
    depth_silhouette[:, 1] = 1.0
    depth_silhouette[:, 2] = depth_z_sq.squeeze(-1)
    
    return depth_silhouette

def transformed_params2depthplussilhouette(params, w2c, transformed_pts):
    rendervar = {
        'means3D': transformed_pts,
        'colors_precomp': get_depth_and_silhouette(transformed_pts, w2c),
        'rotations': F.normalize(params['unnorm_rotations']),
        'opacities': torch.sigmoid(params['logit_opacities']),
        'scales': torch.exp(torch.tile(params['log_scales'], (1, 3))),
        'means2D': torch.zeros_like(params['means3D'], requires_grad=True, device="cuda") + 0
    }
    return rendervar

```

So, by calling `get_depth_and_silhouette(transformed_pts, w2c)`, you do `pts_in_cam = (w2c @ pts4.transpose(0, 1)).transpose(0, 1)`. What does this mean? Didn't we have just transformed point_cloud from world frame to camera frame? 

I know the w2c in this senerio is Identity matrix, so it doesn't make any diffirence. 
But it looks weird to me, it doesn't make any sense to me to transform points from world frame to camera frame twice. So I am wondering if this is a historical issue (somebody forget to change the code) or I am misunderstanding?

jeezrick, 2024-01-31:
@Nik-V9 Hi,  I also noticed that if the input images are not as continuous (for example, with 360 images and an angular difference of 1 degree between each, exactly forming a complete circle), the tracking loss is much larger than that obtained from the Replica demo, reaching up to 18-19 cm.

Nik-V9, 2024-02-03:
Hi @jeezrick, Thanks for your question!

- Re. LangSplat: You are correct. I haven't taken a close look at what they do precisely. Based on a quick look at their paper and appendix, they use the rendered image embeddings to find the objects of interest, as you suggested. Their scheme seems similar to LeRF (https://www.lerf.io/), which might have more details on this.
- Please see this issue for your question regarding the additional w2c: https://github.com/spla-tam/SplaTAM/issues/26
- In the current version of SplaTAM, the number of tracking iterations directly correlates to the amount of motion observed. If the images have significant motion between them you would need more tracking iterations (maybe something similar to the ScanNet++ configs). Please see this issue for more insights: https://github.com/spla-tam/SplaTAM/issues/14#issuecomment-1847467575

jeezrick, 2024-02-19:
@Nik-V9 
Hello, here are my latest experimental results:
I found that during the tracking optimization process, the tracking loss keeps increasing, like '18000 - 41000 - 53000 - ...'. However, when running the official replica demo, the tracking loss remains stable. 
To address this issue, I've tried adjusting the learning rate (switching lrs to the lrs in the ScanNet++ / iPhone capture config) or increasing tracking iterations, but it didn't resolve the problem (increasing tracking iterations improved the results, but didn't stabilize the tracking loss). 
I suspect the issue may lie in the way gaussians are added(in add_new_gaussians function). You mentioned in this issue(https://github.com/spla-tam/SplaTAM/issues/38#issuecomment-1869510218) that you've tried using the 3DGS official prune method but ultimately abandoned it. Could you shed some light on why the current method of adding gaussians was chosen and possibly suggest modifications?

Also, when I use the gt poses(which I checked for many times), the image get blurred very quick. Which also makes me suspect there might be some problems in add new gaussians.

Nik-V9, 2024-03-14:
Hi @jeezrick, Thanks for these experiments.

Some sort of pruning would be effective when ground-truth depth can't be trusted.

I think the problem you are facing, even with gt poses, is incorrect ground truth. I've also observed on my end that the iPhone depth introduces a lot of artifacts. We are working on reducing SplaTAM's reliance on good depth.

Another cool and robust way to use iPhone RGB-D data would be something like this: https://zju3dv.github.io/tof_slam/

closed 2024-03-14 by Nik-V9