spla-tam/SplaTAM #65: What could be the reason that the results from running my own data aren't good?

wangjinhoon, 2024-01-31:
The image below shows the result generated from 500 pieces of data created with an iPhone. Despite inputting gt_data with poses, the results are good for the wall but not for objects inside the room like desks. I'm curious about your thoughts on why this might be the case.
[image]

Nik-V9, 2024-02-03:
Hi, there could be many reasons. My initial guess is the noise and incorrect maximum depth from the iPhone. Could you share this data with us?

Also, I'll shortly be pushing code for Gaussian Splatting with iPhone data. That should provide further insight into why things are failing.

wangjinhoon, 2024-02-05:
Hello, thank you for responding. Below is the data I used. Could there have been an issue with the way I generated my data?

https://drive.google.com/file/d/1qjMWTfspVbYGDg-pZ9NDDw_WerZRyxAF/view?usp=sharing

Nik-V9, 2024-03-14:
Hi, I tried doing large-scale reconstructions similar to your shared data. I've observed that the iPhone data has a lot of artifacts and incorrect depth for scenes with a scale greater than 5 meters.

We will have a follow-up fixing the reliance of SplaTAM on good depth.

wangjinhoon, 2024-03-18:
Thank you for your concern!

closed 2024-03-18 by wangjinhoon