spla-tam/SplaTAM #66: Errors while trying to run in docker

Kelebrimbor97, 2024-02-03:
Hi,

I am trying to run this repo using the instructions provided on docker. The first issue, is that in the documentation ptovided, the sequence of commands that actually worked for me were as shown below:

```bash
docker pull nkeetha/splatam:v1
bash bash_scripts/start_docker.bash
cd /SplaTAM/
apt-get install python3-virtualenv
pip install virtualenv
mkdir venv
cd venv
virtualenv --system-site-packages splatam
source ./splatam/bin/activate
cd ..
pip install -r venv_requirements.txt
```

As you may notice the only significant changes are in line 2 and 4.

Next issue occurs when executing the final command. The requirements installation runs into an error as shown below.

```bash
Looking in indexes: https://pypi.org/simple, https://pypi.ngc.nvidia.com
Collecting git+https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth.git@cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110 (from -r venv_requirements.txt (line 13))
  Cloning https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth.git (to revision cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110) to /tmp/pip-req-build-9zjuszr2
  Running command git clone -q https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth.git /tmp/pip-req-build-9zjuszr2
  Running command git submodule update --init --recursive -q
    ERROR: Command errored out with exit status 1:
     command: /SplaTAM/venv/splatam/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-req-build-9zjuszr2/setup.py'"'"'; __file__='"'"'/tmp/pip-req-build-9zjuszr2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-req-build-9zjuszr2/pip-egg-info
         cwd: /tmp/pip-req-build-9zjuszr2/
    Complete output (11 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-req-build-9zjuszr2/setup.py", line 13, in <module>
        from torch.utils.cpp_extension import CUDAExtension, BuildExtension
      File "/usr/local/lib/python3.8/dist-packages/torch/__init__.py", line 191, in <module>
        _load_global_deps()
      File "/usr/local/lib/python3.8/dist-packages/torch/__init__.py", line 153, in _load_global_deps
        ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
      File "/usr/lib/python3.8/ctypes/__init__.py", line 373, in __init__
        self._handle = _dlopen(self._name, mode)
    OSError: /opt/hpcx/ompi/lib/libmpi.so.40: undefined symbol: opal_hwloc201_hwloc_get_type_depth
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

It is here where I am stuck. How do we proceed after this?

coarsedamavand, 2024-02-21:
@Kelebrimbor97  were you able to find a solution to this?

Kelebrimbor97, 2024-02-28:
> @Kelebrimbor97 were you able to find a solution to this?

I just cloned the repo separately, but then I ran into the same error as #69 after it was 49% done.

Nik-V9, 2024-03-15:
Hi, Thanks for flagging this. We will look into our docker installation instructions @JayKarhade.

For the second issue, the offline version of NeRFCapture is broken. Please see the updated comment on #69.

closed 2024-03-15 by Nik-V9