spla-tam/SplaTAM #67: Question about 'forgetting' or 're-optimization'

lif314, 2024-02-04:
I noticed that during the training process, some frames are initially optimized very well, with PSNR values exceeding 30. However, when I perform evaluation after the entire map optimization process, the rendering results for the initial frames are not satisfactory. Is there any good way to alleviate this issue?

[image]

closed 2024-03-04 by lif314

ljjTYJR, 2024-03-14:
I think it is a trade-off between the efficiency and the accuray.
One option might be enlarging the reply buffer;
The other option would be adding some regularization terms to regularize the optimization of Gaussian parameters.

Nik-V9, 2024-03-15:
I believe that enlarging the mapping window and some updates to our keyframing similar to this paper (https://arxiv.org/abs/2402.03246) should help.