spla-tam/SplaTAM #68: How to draw the Gaussian geometry in the image?

mingqiJ, 2024-02-04:
Is this support drawing the Gaussian geometry in the image? The rasterization engine you used for 3D gaussian Splatting.

Nik-V9, 2024-03-14:
Hi, Yes, you can visualize the Gaussian Geometry in the image by hardcoding the opacity of all Gaussians to 1: https://github.com/spla-tam/SplaTAM/blob/7959433517243b84f8f1b819f8aafb0501edcf9e/viz_scripts/final_recon.py#L79

We will try to add a config in our visualizer.

mingqiJ, 2024-03-20:
Hi, very thanks for your help. I have tried it. Setting all opacities to 1. But it can not show something like this image:
[image]

Do you have any idea about how to show the geometry clear like above image?

Nik-V9, 2024-03-20:
Can you share what you are getting and what exact changes you made?