spla-tam/SplaTAM #71: Rendering depth < 0.2 m

eriksandstroem, 2024-02-09:
Hi,
Thanks a lot for providing this great work open source. 

I find that the current pipeline cannot render depths if they are smaller than 0.2 m. As a toy example, only studying mapping of the first frame: if the frame depth is constant 0.2 m, neither depth nor color can be rendered. If the depth is constant and 0.2001 it works as expected. I have tried various settings, and my conclusion is that it appears to be a global parameter setting somewhere, but I cannot find it sadly.

Do you happen to have any insights on this?

Best,
Erik

closed 2024-02-23 by eriksandstroem

jeezrick, 2024-02-23:
> It is here: https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/blob/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110/cuda_rasterizer/auxiliary.h#L154

Cool! Thanks.

Nik-V9, 2024-03-14:
Thanks for looking into this!

Buffyqsf, 2024-04-16:
> It is here: https://github.com/JonathonLuiten/diff-gaussian-rasterization-w-depth/blob/cb65e4b86bc3bd8ed42174b72a62e8d3a3a71110/cuda_rasterizer/auxiliary.h#L154

Thanks for your finding! I believe that's the problem of my code. I changed the value and re-compile it but it still didn't work with my own data. Have you tried changing the code? I'm wondering the reason why it's not working.