spla-tam/SplaTAM #75: Questions about gaussian params when tracking

Buffyqsf, 2024-02-22:
As the code below in utils.slam_helpers, when setting gaussians_grad false, only params['means3D'] are detached, but other params about gaussians can still be optimized? If so, it's not consistent with what's said in the paper
[image]

[image]

jeezrick, 2024-02-23:
It's about the rationale of autograd. When calculating transformed_pts, it only used cam_rot, cam_trans and means3D. So, you only need to decide if you want to involve these three variables into the grad tracking process. 

Like when you do tracking, when backward the grad, the grad will flow from transformed_pts to (cam_rot, cam_trans and means3D), the other Gaussian parameters like color and opacity wasn't even invovled, so they won't get optimized, as they are't part of the compute.

Buffyqsf, 2024-02-23:
> It's about the rationale of autograd. When calculating transformed_pts, it only used cam_rot, cam_trans and means3D. So, you only need to decide if you want to involve these three variables into the [...]

Thanks for your answering! I think I know the answer, but it's not like what you said(wasn't invovled). I think the reason is that we set learning rate to zero like below. 
[image]

closed 2024-02-23 by Buffyqsf

jeezrick, 2024-02-23:
> Thanks for your answering! I think I know the answer, but it's not like what you said(wasn't invovled). I think the reason is that we set learning rate to zero like below. [...]

I think both reasons are correct. with detach(), means3D won't get any gradients during tracking; Also just set lr to 0, even it has gradient, it won't get updated. 

So, both part make sense. 

Also, if the lrs setting is the whole reason, why they write code like 'params['means3D'].detach()', I think you can verify this by debug the code: you can check that in every iteration in tracking, the gradient if params['means3D'] is None, so even if you set lrs['means3D'] to 0.1, it still work.

So, I guess Im saying that both way works.

Nik-V9, 2024-03-14:
I agree that this was an oversight on our end, however, the 0 learning rate prevents any bugs :)