spla-tam/SplaTAM #79: I could not run the package

czeyveli1, 2024-02-27:
Hello everyone, this is my first time experience about using conda and git package, so I have a trouble about running file. First of all, I setup anaconda from official page on my Ubuntu 20.04 system and I executed these command properly.

```
conda create -n splatam python=3.10
conda activate splatam
conda install -c "nvidia/label/cuda-11.6.0" cuda-toolkit
conda install pytorch==1.12.1 torchvision==0.13.1 torchaudio==0.12.1 cudatoolkit=11.6 -c pytorch -c conda-forge
pip install -r requirements.txt
```

After that, I cloned splaTAM package to my computer and tried to run splatam.py file but I get this error. 
```
System Paths:
/home/cz/anaconda3/SplaTAM
/home/cz/anaconda3/SplaTAM/scripts
/home/cz/anaconda3/envs/splatam/lib/python310.zip
/home/cz/anaconda3/envs/splatam/lib/python3.10
/home/cz/anaconda3/envs/splatam/lib/python3.10/lib-dynload
/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages
Traceback (most recent call last):
  File "/home/cz/anaconda3/SplaTAM/scripts/splatam.py", line 16, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2' 
```

I guess it is about the lack of library file but I think I don't have to add libraries one by one. I made a mistake somewhere while uploading but I couldn't find what my mistake was.
### Could you give me more explanation about how can I setup all packages?

Nik-V9, 2024-03-14:
Hi, Thanks for trying our code!

This seems strange. The instructions you ran should have installed cv2. Did you make sure all the commands were run one by one?

Also you should clone the SplaTAM code and change your directory to it before running the installation commands.

Nik-V9, 2024-03-14:
I see your follow-up issue. Looks like this issue is resolved :)

closed 2024-03-14 by Nik-V9