spla-tam/SplaTAM #81: Building wheel for diff-gaussian-rasterization (setup.py) ... error

czeyveli1, 2024-02-28:
Hello dear friends,
When I try to install requirements.txt using `pip install -r requirements.txt` command, I get an error like;
```
Building wheels for collected packages: diff-gaussian-rasterization
  Building wheel for diff-gaussian-rasterization (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [60 lines of output]
      running bdist_wheel
      /home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py:411: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
        warnings.warn(msg.format('we could not find ninja.'))
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-310
      creating build/lib.linux-x86_64-cpython-310/diff_gaussian_rasterization
      copying diff_gaussian_rasterization/__init__.py -> build/lib.linux-x86_64-cpython-310/diff_gaussian_rasterization
      running build_ext
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/tmp/pip-req-build-zunyy40x/setup.py", line 17, in <module>
          setup(
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/wheel/bdist_wheel.py", line 364, in run
          self.run_command("build")
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/command/build.py", line 131, in run
          self.run_command(cmd_name)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/command/build_ext.py", line 88, in run
          _build_ext.run(self)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 345, in run
          self.build_extensions()
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 420, in build_extensions
          compiler_name, compiler_version = self._check_abi()
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 791, in _check_abi
          _, version = get_compiler_abi_compatibility_and_version(compiler)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 324, in get_compiler_abi_compatibility_and_version
          if not check_compiler_ok_for_platform(compiler):
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 280, in check_compiler_ok_for_platform
          which = subprocess.check_output(['which', compiler], stderr=subprocess.STDOUT)
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/subprocess.py", line 420, in check_output
          return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
        File "/home/cz/anaconda3/envs/splatam/lib/python3.10/subprocess.py", line 524, in run
          raise CalledProcessError(retcode, process.args,
      subprocess.CalledProcessError: Command '['which', 'g++']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for diff-gaussian-rasterization
  Running setup.py clean for diff-gaussian-rasterization
Failed to build diff-gaussian-rasterization
ERROR: Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-based projects
```
And also I cannot execute anything because of gaussian-rasterization.

LinijaVP, 2024-02-28:
I solved the problem by using pip install instead of conda install for the torch packages.

czeyveli1, 2024-02-29:
Same error..

czeyveli1, 2024-02-29:
Okey, I tried to clone diff-gaussian-rasterization-w-depth.git file manually and put it inside the SplaTAM folder manually, and also I tried to install "setup.py" file using `python setup.py install` and get an error as;
```
File "/home/cz/anaconda3/envs/splatam/lib/python3.10/subprocess.py", line 421, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/home/cz/anaconda3/envs/splatam/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['which', 'g++']' returned non-zero exit status 1.
```
I do not now is there any other way to solve the problem?

czeyveli1, 2024-03-04:
Finally, I have solved my problem using the issue **#24** .  As mentioned in the issue, the problem is the version of gcc&g++ version is incompatable with CUDA. I fixed the problem using `conda install gxx_linux-64=10` command.

closed 2024-03-04 by czeyveli1