spla-tam/SplaTAM #90: Is it applicable for facial expression recognition tasks

Mysterious-handsome-man, 2024-03-11:
Impressive work! May I ask if this work can be used for obtaining dynamic 3D facial videos (RGBD video) from monocular video input (RGB video), and using the output RGB and 3D information for facial expression recognition tasks?

Nik-V9, 2024-03-14:
Hi, Thanks for your interest in our work!

In its current form, SplaTAM only works for static scenes. We might have follow-ups in the future that would work on dynamic monocular videos.

For the application you mentioned, these papers might be relevant (although they require camera poses):
- https://shunsukesaito.github.io/rgca/
- https://animatable-gaussians.github.io/
- https://github.com/MrNeRF/awesome-3D-gaussian-splatting?tab=readme-ov-file#avatars

Mysterious-handsome-man, 2024-03-15:
Thank you very much for your recommendation! It's very inspiring to me. I will also continue to follow your work!

closed 2024-03-17 by Nik-V9