spla-tam/SplaTAM #91: EGL error when trying to use your code to get the undistort depth from Scanet ++

Bensong0506, 2024-03-11:
Hi, 

I managed to install the renderpy and tried to get the undistort depth from the repo you provide, but I get the following error. I am using ubuntu 20.04 btw.

File "/shared/scannetpp/common/render.py", line 47, in main
    render_engine = renderpy.Render()
RuntimeError: EGL error at eglInitializewith error code: 12296

Nik-V9, 2024-03-14:
Hi, Thanks for trying our code! 

I think this is not a problem with our modifications to the original scannetpp code. Are you able to render distorted RGB or depth?

I would recommend opening an issue on the scannetpp repo: https://github.com/scannetpp/scannetpp

XiangyuSu611, 2024-03-25:
@Bensong0506 hi,

 have you sloved this problem?