spla-tam/SplaTAM #93: After running post_splatam_opt.py for densification, the newly generated npz file has worse performance compared to splatam.py

LemonSoda-RPG, 2024-03-12:
First of all, thank you very much for your work.
I followed the steps in the readme file and ran the following code step by step, and I used the replica dataset.

1. `python scripts/splatam.py configs/replica/splatam.py`
2. `python scripts/post_splatam_opt.py configs/replica/post_splatam_opt.py`

This image is the result after running only `splatam.py`.
[image]
This image is the result after running `post_splatam_opt.py` following the execution of `splatam.py`.
[image]

I attempted to disable densification in the configuration file of post_splatam_opt, but the results remained the same.

Nik-V9, 2024-03-14:
Hi, definitely something seems wrong here. @JayKarhade will get back to you.

JayKarhade, 2024-03-20:
Hi @LemonSoda-RPG i got some time to check this - it seems to work fine on my end.

Could you share your config files and post_splatam_opt.py file? I'd like to check if there is any mismatch.

LemonSoda-RPG, 2024-03-21:
> Hi @LemonSoda-RPG i got some time to check this - it seems to work fine on my end.
> 
> Could you share your config files and post_splatam_opt.py file? I'd like to check if there is any mismatch.

Thank you very much for your reply！
To verify again, I used the latest version of the code, the replica dataset, and executed the following command again.

Compared with the latest version of the code in the github repository, I only made two modifications. When running splatam, I set in the configuration file to use the real pose, not track, and closed wandb. **Other than that, there is no Make any changes.**

1. `python scripts/splatam.py  configs/replica/splatam.py`
2. `python scripts/post_splatam_opt.py configs/replica/post_splatam_opt.py`

But after running post_splatam_opt.py, I still got worse results.

Nik-V9, 2024-06-08:
Hi, can you please share the complete config files so that we can try running this on our end?

I'm not sure why using ground truth poses would cause this bug. Is the `use_gt_poses` config set consistently across both files? Do you still see the bug when you try the default configs?