spla-tam/SplaTAM #103: Error when running SplaTAM on Scannet++ dataset

TianyuHuang-000, 2024-04-05:
Thank you for your great work !

I followed the steps in the readme file, first undistorted the dataset, then ran the following code and get the following error
Any help and suggestions would be appreciated.

python scripts/splatam.py configs/scannetpp/splatam.py

[image]

TianyuHuang-000, 2024-04-08:
Need to render the depth first, then distort the images

closed 2024-04-08 by TianyuHuang-000

chanwendy, 2024-04-17:
Hi, have you solved this problem, I have also encountered this problem and would like to ask how you solved it?

chanwendy, 2024-04-17:
> Thank you for your great work !
> 
> I followed the steps in the readme file, first undistorted the dataset, then ran the following code and get the following error Any help and suggestions would be appreciated.
> 
> python scripts/splatam.py configs/scannetpp/splatam.py
> 
> ![截屏2024-04-05 上午11 45 59](https://private-user-images.githubusercontent.com/88012819/319815359-080add88-c4bb-476d-8186-f19f75898ebf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTMzMzU1OTIsIm5iZiI6MTcxMzMzNTI5MiwicGF0aCI6Ii84ODAxMjgxOS8zMTk4MTUzNTktMDgwYWRkODgtYzRiYi00NzZkLTgxODYtZjE5Zjc1ODk4ZWJmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MTclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDE3VDA2MjgxMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRkM2IzYTA5Yjg5YzU2YTQ1NTMyMWNjYjJkZjkwZTQxYWE5OGQzMmI3NWJlZDY0Mjc5ZTQ0OTBiMGY1YmE4M2MmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vyHKhWdI5jYfONhNxGUhjZXpfE-amfeg-0aVMVtk_ww)

Hi, have you solved this problem, I have also encountered this problem and would like to ask how you solved it?

BINOFL, 2024-05-23:
> Hi, have you solved this problem, I have also encountered this problem and would like to ask how you solved it?

error occurs because the scannet++ dataset dslr depth render is not performed