spla-tam/SplaTAM #104: Question about render hole in my own data

Buffyqsf, 2024-04-09:
I got a hole in my own captured data as below, I've tried to find out the reason but failed. 
[image]
And I found there is Gaussian, because when I change the view camera's z position, the hole dispeared like below. 
Wondering if you have any experience in such situation.
[image]

Buffyqsf, 2024-04-09:
I found there seems to have a crop depth min(near 0.2), but I can't find any code can change this(I've tried to adjust near and far or fov in my view camera)

Nik-V9, 2024-04-16:
Hi, please refer to this comment: https://github.com/spla-tam/SplaTAM/issues/71#issuecomment-1960921068

closed 2024-04-16 by Nik-V9