spla-tam/SplaTAM #106: `RuntimeError: numel: integer multiplication overflow`

kevintsq, 2024-04-18:
When running SplaTAM on the same dataset, the following error sometimes happens. Why?
```sh
    im, radius, _, = Renderer(raster_settings=curr_data['cam'])(**rendervar)
  File "anaconda3/envs/splatam/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1130, in _call_impl
    return forward_call(*input, **kwargs)
  File "anaconda3/envs/splatam/lib/python3.10/site-packages/diff_gaussian_rasterization/__init__.py", line 186, in forward
    return rasterize_gaussians(
  File anaconda3/envs/splatam/lib/python3.10/site-packages/diff_gaussian_rasterization/__init__.py", line 28, in rasterize_gaussians
    return _RasterizeGaussians.apply(
  File "anaconda3/envs/splatam/lib/python3.10/site-packages/diff_gaussian_rasterization/__init__.py", line 79, in forward
    num_rendered, color, radii, geomBuffer, binningBuffer, imgBuffer, depth = _C.rasterize_gaussians(*args)
RuntimeError: numel: integer multiplication overflow
```

gwilczynski95, 2024-06-05:
Same problem here. Not in this project but I'm also using rasterize_gaussian function and I get this error

Nik-V9, 2024-06-08:
Hi, this seems to be a problem with the 3DGS rasterizer.

I would recommend checking the issues on the official repo: https://github.com/graphdeco-inria/gaussian-splatting

sherryQQ1996, 2024-06-11:
> Hi, this seems to be a problem with the 3DGS rasterizer.
> 
> I would recommend checking the issues on the official repo: https://github.com/graphdeco-inria/gaussian-splatting

Hi, I also meet this problem