spla-tam/SplaTAM #111: Code fusion

suvzzc, 2024-05-22:
Thank you for your excellent work. I would like to ask why the pose read from the dataset "groundtruth. txt" is used? Isn't it position estimation? Why is the pose from the dataset used?
[image]

Nik-V9, 2024-06-08:
Hi, Thanks for your interest!

We configure our dataset to return poses relative to the first frame. Hence, the first camera pose that we load with dataset[0] is always identity: https://github.com/spla-tam/SplaTAM/blob/1a5d407c3a145f69691d49267411cb65e5d5c992/scripts/splatam.py#L529

However, when you want to use absolute poses for SLAM (which is conventionally not preferred and done), the code using dataset[0] supports this (as opposed to just hardcoding the first pose to identity).

Also, these additional details about pose estimation might be relevant:
1. https://github.com/spla-tam/SplaTAM/issues/30#issuecomment-1853483102
2. https://github.com/spla-tam/SplaTAM/issues/28

suvzzc, 2024-06-14:
> Hi, Thanks for your interest!
> 
> We configure our dataset to return poses relative to the first frame. Hence, the first camera pose that we load with dataset[0] is always identity:
> 
> https://github.com/spla-tam/SplaTAM/blob/1a5d407c3a145f69691d49267411cb65e5d5c992/scripts/splatam.py#L529
> 
> However, when you want to use absolute poses for SLAM (which is conventionally not preferred and done), the code using dataset[0] supports this (as opposed to just hardcoding the first pose to identity).
> 
> Also, these additional details about pose estimation might be relevant:
> 
>     1. [a small question about ref_w2c #30 (comment)](https://github.com/spla-tam/SplaTAM/issues/30#issuecomment-1853483102)
> 
>     2. [Q about pose estimation #28](https://github.com/spla-tam/SplaTAM/issues/28)
Thank you for your reply. I re-read the code and realized that I misunderstood.  Here gt_pose uses relative pose and it is not used in the trace section.

closed 2024-06-14 by suvzzc