spla-tam/SplaTAM #112: Bug Report: Scannet Dataset Config

Ramseyous0109, 2024-05-29:
Hi, I've found a small bug on the config of Scannet Dataset In `configs/data/scannet.yaml`

```yaml
dataset_name: 'scannet'
camera_params:
  image_height: 968
  image_width: 1296
  fx: 1169.621094
  fy: 1167.105103
  cx: 646.295044
  cy: 489.927032
  png_depth_scale: 1000.0
```

This intrinsic data refers to the original color images' shape. However, since the color image should be cropped to `640 × 480` to fit the size of the depth data, I think the intrinsics should also be changed [as the config in NICE-SLAM ](https://github.com/cvg/nice-slam/blob/master/configs/ScanNet/scannet.yaml)to get correct results.

```yaml
cam: 
  H: 480 
  W: 640
  fx: 577.590698
  fy: 578.729797
  cx: 318.905426
  cy: 242.683609
  png_depth_scale: 1000. #for depth image in png format
```

I wonder if I get this config right. Looking forward to your reply@

Nik-V9, 2024-06-08:
Hi, since our dataloader takes care of resizing intrinsic when resizing the RGB image, we provide the intrinsic parameters for the full-resolution images in the dataset config file:
https://github.com/spla-tam/SplaTAM/blob/1a5d407c3a145f69691d49267411cb65e5d5c992/datasets/gradslam_datasets/basedataset.py#L318

Ramseyous0109, 2024-06-15:
Thanks for your reply! It's clear now

closed 2024-06-15 by Nik-V9