spla-tam/SplaTAM #113: Need to run for a different resolution

sumitanilkshirsagar, 2024-05-31:
Hi, 
I am working on a underwater 3D reconstruction project, and wanted to first establish a backbone 3D reconstruction library and then build on top of it. HengyiWang(Co-SLAM ) suggested thaT your code would work better for under water 3D purposes.
I wanted to check how I can build a dataset and config, I am using a stereo camera (ZED camera from stereolabs) with a resolution of HD2K ( 2208x1242)

sumitanilkshirsagar, 2024-05-31:
1) I tried to resize my rgb,depth images to 640x480 and placing them in the TUM dataset folder. 
Error:

SplaTAM/scripts/splatam.py", line 107, in get_pointcloud
    point_cld = [torch.cat](http://torch.cat/)((pts, cols), -1)
RuntimeError: Sizes of tensors must match except in dimension 1. Expected size 307200 but got size 409600 for tensor number 1 in the list.

Can you help me with what these tensors denote, and where is mismatch coming from?

Nik-V9, 2024-06-08:
Hi, Thanks for your interest in our work!

You will need to implement your own dataloader using the ones present here as reference: https://github.com/spla-tam/SplaTAM/tree/main/datasets/gradslam_datasets

You will also need a corresponding config file for the data, which contains the intrinsic & depth PNG scale information (pixel value to depth mapping): https://github.com/spla-tam/SplaTAM/tree/main/configs/data

Regarding the issue you are facing, there might be a mismatch between your RGB & depth image size.

ShreyaIAB, 2024-08-20:
@sumitanilkshirsagar were u able to solve this and load ur own custom data with your camera?