spla-tam/SplaTAM #119: diff-gaussian-rasterization error

whwh747, 2024-06-28:
Hello,
Thank you very much for sharing such excellent work. However, I find it very difficult to set up the environment. My CUDA version is 11.6, and I followed the instructions in the README for the rest of the setup. However, I encounter errors when installing the submodules using pip install .. The error message is as follows:
Processing /disk2/wh/code/SplaTAM_wh/diff-gaussian-rasterization-w-depth
  Preparing metadata (setup.py) ... done
Building wheels for collected packages: diff-gaussian-rasterization
  Building wheel for diff-gaussian-rasterization (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py bdist_wheel did not run successfully.
  │ exit code: 1
  ╰─> [70 lines of output]
      /disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/cuda/__init__.py:83: UserWarning: CUDA initialization: Unexpected error from cudaGetDeviceCount(). Did you run some cuda functions before calling NumCudaDevices() that might have already set an error? Error 804: forward compatibility was attempted on non supported HW (Triggered internally at  /opt/conda/conda-bld/pytorch_1659484803030/work/c10/cuda/CUDAFunctions.cpp:109.)
        return torch._C._cuda_getDeviceCount() > 0
      No CUDA runtime is found, using CUDA_HOME='/disk2/wh/anaconda3/envs/splatam'
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build/lib.linux-x86_64-cpython-310
      creating build/lib.linux-x86_64-cpython-310/diff_gaussian_rasterization
      copying diff_gaussian_rasterization/__init__.py -> build/lib.linux-x86_64-cpython-310/diff_gaussian_rasterization
      running build_ext
      /disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py:820: UserWarning: There are no /usr/bin/g++-10 version bounds defined for CUDA version 11.6
        warnings.warn(f'There are no {compiler_name} version bounds defined for CUDA version {cuda_str_version}')
      building 'diff_gaussian_rasterization._C' extension
      creating /disk2/wh/code/SplaTAM_wh/diff-gaussian-rasterization-w-depth/build/temp.linux-x86_64-cpython-310
      creating /disk2/wh/code/SplaTAM_wh/diff-gaussian-rasterization-w-depth/build/temp.linux-x86_64-cpython-310/cuda_rasterizer
      Traceback (most recent call last):
        File "<string>", line 2, in <module>
        File "<pip-setuptools-caller>", line 34, in <module>
        File "/disk2/wh/code/SplaTAM_wh/diff-gaussian-rasterization-w-depth/setup.py", line 17, in <module>
          setup(
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/wheel/bdist_wheel.py", line 368, in run
          self.run_command("build")
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/command/build.py", line 131, in run
          self.run_command(cmd_name)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/dist.py", line 989, in run_command
          super().run_command(command)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/command/build_ext.py", line 88, in run
          _build_ext.run(self)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 345, in run
          self.build_extensions()
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 765, in build_extensions
          build_ext.build_extensions(self)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 467, in build_extensions
          self._build_extensions_serial()
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 493, in _build_extensions_serial
          self.build_extension(ext)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/command/build_ext.py", line 249, in build_extension
          _build_ext.build_extension(self, ext)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/setuptools/_distutils/command/build_ext.py", line 548, in build_extension
          objects = self.compiler.compile(
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 581, in unix_wrap_ninja_compile
          cuda_post_cflags = unix_cuda_flags(cuda_post_cflags)
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 480, in unix_cuda_flags
          cflags + _get_cuda_arch_flags(cflags))
        File "/disk2/wh/anaconda3/envs/splatam/lib/python3.10/site-packages/torch/utils/cpp_extension.py", line 1694, in _get_cuda_arch_flags
          arch_list[-1] += '+PTX'
      IndexError: list index out of range
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for diff-gaussian-rasterization
  Running setup.py clean for diff-gaussian-rasterization
Failed to build diff-gaussian-rasterization
ERROR: Could not build wheels for diff-gaussian-rasterization, which is required to install pyproject.toml-based projects

JzHuai0108, 2024-07-18:
I think this is related to [issue](https://github.com/spla-tam/SplaTAM/pull/24) which has been solved.

whwh747, 2024-07-18:
thank you！ @JzHuai0108

closed 2024-07-18 by whwh747