spla-tam/SplaTAM #125: D435i real-time or offline slam?

Tanfuchy, 2024-07-09:
If I connect the D435i camera to my computer by usb, how can I use this camera and your project for real-time SLAM? Or can you only support SLAM that uses a camera to shoot a piece of data first and then go offline? Could you please give more detailed steps and which config file to use (iphone? )

Nik-V9, 2024-08-29:
Hi, Thanks for your interest!

Unfortunately, we don't currently support off-the-shelf stereo cameras.

To run our iPhone-based demo, you can refer to these instructions: https://github.com/spla-tam/SplaTAM?tab=readme-ov-file#demo

saracarles, 2025-05-14:
Hello, I'm trying to use this code too and I had the same question I found here! I wanted to do the online demo with my iphone 13 but sadly it does not support depth. My alternative was to connect my D435i camera to do online but I see here that it is not compatible right now. 
Can I do something to modify the code so the online demo is compatible with my D435i camera? Where should I modify the code?
I'm having trouble finding in the code where the camera of the iphone is called. I thought if I found where it is called and modify the segment to call my D435i camera I could make it compatible.
Thanks!