spla-tam/SplaTAM #131: question about the first pose

mah-asd, 2024-09-09:
Hi, thank you for the great work. 

I have a question about the first pose which should be identity, if I transform the first pose in dataset, then the splatam uses that first frame pose to do tracking for the rest of frame relative to that. But by changing the first frame to a non identity matrix, the result is very bad. I think I'm missing something here.

Nik-V9, 2024-09-13:
Hi, Thanks for your interest in our work! Can you elaborate further?

I think this behavior might have to do with the stability of optimization world frame poses as opposed to relative poses.

mah-asd, 2024-09-13:
It seems like even if the relative_pose in dataset is set to False, the initialization for camera poses in params and also the gaussians translations and rotations are defined to have the first pose as identity. I manually changed them according to gt_pose for first frame, but I think some where in eval function also needs some adjustment.

mah-asd, 2024-09-30:
[image]

This is the result when I only change relative_pose variable in loading dataset to False

Nik-V9, 2024-09-30:
Hi, I don't quite follow. This frame looks correct to me. Are you indicating that new gaussian aren't being added?

mah-asd, 2024-09-30:
Thanks for your reply, how do you interpret the PSNR 16.22?

mah-asd, 2024-10-01:
I mean how do I fix the poses for eval part? it seems like there is a transformation mismatch.