spla-tam/SplaTAM #132: Problems encountered during visualization of results

lxin98, 2024-09-12:
Thank you for your work. I am running Python viz_stcripts/final_decon.exe configs/realsense/splatam.py
When coding, there was an error called 'Process finished with exit code 139' (interrupted by signal 11: SIGSEGV). I debugged and found that it was at line 202 where frustum=o3d. geometry LineSet.creat_camera_visualization (w, h, k, allw_2cs [i_t], frustumsize) encountered an error. How can I resolve it

lxin98, 2024-09-12:
[image]

Nik-V9, 2024-09-13:
Hi, Thanks for trying out our code!

Do you have the right version  of Open3D installed?

https://github.com/spla-tam/SplaTAM/blob/da6bbcd24c248dc884ac7f49d62e91b841b26ccc/requirements.txt#L11

lxin98, 2024-09-13:
yes ，I installed the right version0.16.0，I am going to create a new conda enviroment。I plan to create a new conda environment to see if it can run @Nik-V9 ，thank you for you help