spla-tam/SplaTAM #138: Question

AIerOK, 2024-12-05:
Traceback (most recent call last):
  File "/home/xuxianghong/下载/SplaTAM-main(1)/SplaTAM-main/scripts/splatam.py", line 28, in <module>
    from utils.eval_helpers import report_loss, report_progress, eval
  File "/home/xuxianghong/下载/SplaTAM-main(1)/SplaTAM-main/utils/eval_helpers.py", line 10, in <module>
    from utils.recon_helpers import setup_camera
  File "/home/xuxianghong/下载/SplaTAM-main(1)/SplaTAM-main/utils/recon_helpers.py", line 2, in <module>
    from diff_gaussian_rasterization import GaussianRasterizationSettings as Camera
  File "/home/xuxianghong/anaconda3/envs/splatam/lib/python3.10/site-packages/diff_gaussian_rasterization/__init__.py", line 15, in <module>
    from . import _C
ImportError: /home/xuxianghong/anaconda3/envs/splatam/lib/python3.10/site-packages/diff_gaussian_rasterization/_C.cpython-310-x86_64-linux-gnu.so: undefined symbol: _ZNK3c1010TensorImpl18compute_contiguousEv

Kaiqi-Chen, 2025-01-18:
@AIerOK Hi, have you solved this problem?

zhzhanp, 2025-01-24:
I came across the same problem. This is because the version of pytorch does not match the one used when we install diff_gaussian_rasterization. Why this happen? In my case, when perform pip install -r requirement.txt, where we will install torchmetrics, and it automatically upgrade pytorch to the latest version, say 2.xxx, not the one 1.12.xx. So just remove torchmetrics in requirement.txt and install it together with pytorch==1.12.