spla-tam/SplaTAM #140: question about offline demo

Zhaooyy, 2024-12-17:
Does offline demo need  iPhone and PC are connected to the same WiFi network? I don't have such condition, and when i run `bash bash_scripts/nerfcapture.bash configs/iphone/nerfcapture.py`, i stuck on `Waiting for frames...`.

So if the WIFI is a necessery condition to run offline deom? If not, what can i do to finish the experiment?

erliya, 2025-01-22:
同问，感觉离线模式也需要通信的样子

nlc2002, 2025-08-10:
> Does offline demo need iPhone and PC are connected to the same WiFi network? I don't have such condition, and when i run `bash bash_scripts/nerfcapture.bash configs/iphone/nerfcapture.py`, i stuck [...]

Is this problem solved now?